FreeCAD/FreeCAD #11083: [Problem] Rendering issue in SubtractiveHelix

VolkerRaab, 2023-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The rendering of a SubtractiveHelix depends subtly on one parameter of the used sketch to be subtracted. The thread shows a "pimple" in the generated thread.
[image]
Also the smoothness in nearby threads looks very deteriorated.
The "real" problem then results from the fact that also for an export (e.g. as *.AMF) for a 3D-print this error is persistent.
When the diameter of the underlying sketch is changed from 23mm to 22.99 (encircled) everything looks O.K.
[image]

### Full version info

```shell
[code]
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git) AppImage
Build type: Release
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[bug-suspect_Sketch020_23mm-vs-22.9mm.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/12999834/bug-suspect_Sketch020_23mm-vs-22.9mm.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-10-24:
This may just be a recompute issue. With the mentioned constraint set to 23mm, if I recompute the model the problem goes away.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git)
Build type: Release
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
```

VolkerRaab, 2023-10-24:
@Roy-043 
Thanks a lot for having such a quick look at it!
Yet, if I switch back and forth between 23 and something else, the error always reproduces on the 23. However, it disappears even at 23.0001. This is minute and in principle would be a workaround, but doesn't solve the issue - particularly not for unseen bumps :-(
The only difference between your computation and mine seems to be Linux vs. Windows. All software versions are in accordance. (I even re-downloaded my own zip-file ;-) )
I don't know how to detect whether it is an OCC-issue? I'm lost on that: what can I do to check it?

Roy-043, 2023-11-09:
With a retest under Ubuntu I can confirm the problem. It is almost certainly OCC related. For example changing the Angle of Sketch020 (under Attachment) to 15 degrees, with the constraint value kept at 23mm, also solves the problem.

```
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34836 (Git)
Build type: Release
Branch: main
Hash: 862eef57eb4aa042f87d9f19dc768a017b17dae6
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
```

AIRCAP, 2024-10-18:
#17119 can fix this, but only for very large base fuzzy values (min 100 to have a shape that "looks right" , min 1000 for a shape that also is a valid solid that passes BOP checks -- 1000 might be too high for other cases)

AIRCAP, 2024-10-18:
note: using 17119 causes the number of faces on the first (inner) subtractive helix to change, with the effect of breaking sketch attachment. That's because the inner helix is also broken.

AIRCAP, 2024-10-18:
weird thing: the bug only appears if the reference axis for the helix is the sketch normal (which is 0,0,1 - origin 0,0,16) if the reference axis is instead set to the base Z axis (0,0,1, origin 0,0,0) then the bug disappears... why?

PR #17356 (2024-11-11): increase tolerance on helical shapes to avoid crash / broken geometry 
fix #17113
fix #9377
fix #11083

Helical shapes in PartDesign (and to a lesser degree helical paths) suffer from a precision flaw (see #9337) -- cylinders, cones, spheres, boxes, etc... are all "perfect" shapes in OCCT - ruled by exact mathematical formulas where their edges and surfaces are.

But OCCT has no primitive shape for helical shapes - be it 2d and 3d - so helical paths are approximated with bsplines and helicap solids are pipes along these bsplines.

now the issue is this approximation makes errors far in excess  [... cut]
commit 7680872e (2024-11-11): increase tolerance on helical shapes to avoid crash / broken geometry  (#17356)

* increase tolerance on helical shapes to avoid crash

fix #17113
fix #9377
fix #11083

* added test cases for additive and subtractive helix

* fixed whitespace issue, reduced turncount for speed

* relaxed large scale testcases for OCCT 7.3 and below, work around windows CI error #17394

* Further reduce max size in test for buggy Ubuntu OCCT
commit 6d68d222 (2024-11-11): increase tolerance on helical shapes to avoid crash / broken geometry  (#17356)

* increase tolerance on helical shapes to avoid crash

fix #17113
fix #9377
fix #11083

* added test cases for additive and subtractive helix

* fixed whitespace issue, reduced turncount for speed

* relaxed large scale testcases for OCCT 7.3 and below, work around windows CI error #17394

* Further reduce max size in test for buggy Ubuntu OCCT
closed 2024-11-11 by yorikvanhavre