FreeCAD/FreeCAD #11084: [Feature Request] Merging of Part Design and Part

NXTler, 2023-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello all,
since I started learning FreeCAD, I have always wondered why there were both the Part and Part Design Workbench. I'm an experienced CAD user, and it seems kind of an odd choice to do. Adding unnecessary complexity to the whole system.
So I asked the Discord community of their opinion on this topic, and here are some of the thoughts I got:
- Having two Workbenches is quite confusing, especially for newcomers.
- There are many cases where the user is tempted to (or even forced to) use both Workbenches at the same time. Which is really blurring the line between them.
- One of the Workbenches has to be merged into the other. "[...] Either way, they are probably more alike, than they are different."

Of course, this will not be easy to pull off and without a deeper knowledge of FreeCAD I can't even tell where to start at. Hence, I want this to be an open discussion where everyone can leave their thoughts about it.
Many thanks in advance!

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git) AppImage
Build type: Release
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * PieMenu 1.2.7
  * sheetmetal 0.3.1
  * A2plus 0.4.60n
  * QuickMeasure 2022.10.28
  * Curves 0.6.13
  * CfdOF 1.24.5
  * IconThemes
  * fasteners 0.4.67
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

In my opinion, the best approach is trying to merge Part Design into Part.
Part has the benefit of providing some more functionality than Part Design, being able to create disjoint bodies, for example. However, I would like to keep the Part Design's UI/Workflow. With additive and subtractive functions already in toolbar, the way you create chamfers/fillets by selecting edges, the easier creation of sketches and some other things.
This would make Part easier to use for newcomers, as well as keeping the capability to create more complex objects.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-10-25:
Maybe in a first step rename the Part WB to something less confusing for new users. https://forum.freecad.org/viewtopic.php?style=10&t=50256

NXTler, 2023-10-25:
This would definitely be a step in the right direction.

WandererFan, 2023-11-12:
The names suck, but there are valid arguments for having both workbenches.  Brand new users deserve something simpler than either.
 
https://forum.freecad.org/viewtopic.php?t=6082
https://forum.freecad.org/viewtopic.php?t=56055
https://forum.freecad.org/viewtopic.php?t=27436