FreeCAD/FreeCAD #11085: DXF importer error

69Piero, 2023-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

some errors in opening dxf

https://forum.freecad.org/viewtopic.php?t=82023

[DXF.zip](https://github.com/FreeCAD/FreeCAD/files/13025122/DXF.zip)

### Full version info

```shell
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git)
Build type: Release
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * 3DfindIT 1.2.0
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * dxf-library
  * fasteners 0.4.67
  * fasteners.backup1693551082.0742238 0.4.62 (Disabled)
  * freecad.gears 1.0.0
  * QuickMeasure 2022.10.28
  * Silk 0.1.5
  * ThreadProfile 1.89.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-18:
CC @yorikvanhavre

69Piero, 2023-10-18:
the same file converted into Autocad 2000 DXF version give back following errors:
 
1**4:27:38 Traceback (most recent call last):
File "<string>", line 1, in <module>
File "C:\FreeCAD\FreeCAD_weekly-builds-34667-2023-10-15-conda-Windows-x86_64-py310\Mod\Draft\importDXF.py", line 2722, in open
processdxf(doc, filename)
File "C:\FreeCAD\FreeCAD_weekly-builds-34667-2023-10-15-conda-Windows-x86_64-py310\Mod\Draft\importDXF.py", line 2158, in processdxf
color = tuple(dxfColorMap.color_map[layer.color])
<class 'KeyError'>: (-4,)**

Roy-043, 2023-10-18:
#11090 fixes the `<class 'KeyError'>: (-4,)`.

closed 2023-10-19 by yorikvanhavre