FreeCAD/FreeCAD #11087: [Problem] In Techdraw when export to PDF using the 'Print all pages' tool, the style/font changes

DEL-TRONIK, 2023-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

By using the 'Print all pages' tool in the techdraw workshop to export your drawing as a PDF file, when exported, the PDF file does not have the same writing style as on FreeCAD. It looks like he's adding **bold** style to the cartouche writing, but maybe it's something else? perhaps it prints the cartridge twice? But using the 'File->Export PDF' tool works perfectly.

An example of the difference between the 'Print all pages' and 'File->Export PDF' tools:

[image]

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git) AppImage
Build type: Release
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.19.5
  * Defeaturing 1.2.0
  * ThreadProfile 1.89.0
  * fasteners 0.4.66
  * ExplodedAssembly
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-18:
Maybe related to https://github.com/FreeCAD/FreeCAD/issues/10976#issuecomment-1753234693

WandererFan, 2023-10-20:
should be fixed by PR #11125.

luzpaz, 2023-10-21:
Thanks WandererFan!  
@DEL-TRONIK please re-open if necessary.

closed 2023-10-21 by luzpaz