FreeCAD/FreeCAD #11088: [Problem] Techdraw does not export hatchings in PDF format if their value is less than or equal to 0.5

DEL-TRONIK, 2023-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using Techdraw and the hatch tool, when modifying the 'SVG Pattern Scale' property or the 'Hatch Scale' tool when using a section view, hatchings are not displayed when exporting with the 'Export PDF' tool, if the value is less than or equal to 0.5. This works if you leave the default hatch value, or if you set something higher than 0.5. However, it works when exporting with the 'Print all pages' tool.

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git) AppImage
Build type: Release
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.19.5
  * Defeaturing 1.2.0
  * ThreadProfile 1.89.0
  * fasteners 0.4.66
  * ExplodedAssembly
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-10-23:
Should be fixed by PR #11147.

luzpaz, 2023-10-25:
Closing. If this can still be reproduced, please re-open.

closed 2023-10-25 by luzpaz