FreeCAD/FreeCAD #11091: [Problem] PVS-Studio warnings

AndreyKarpov, 2023-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello. I found a few errors in the code. I will be glad if the project becomes a little better.
[FreeCAD and undefined behavior in C++ code: meditation for developers](https://pvs-studio.com/en/blog/posts/cpp/1072/).

### Full version info

```shell
I do not remember. I don't think it matters :)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2023-10-18:
https://github.com/FreeCAD/FreeCAD/pull/11093

wwmayer, 2023-10-19:
> inline uint32 readUint32 ( istream &is )
This is how the new zipios library handles it: https://github.com/Zipios/Zipios/blob/8ff715264283f489e9d0d50d9062b7de5a2114d5/src/zipios_common.cpp#L74

WandererFan, 2023-10-19:
1) QGVPage* QGIView::getQGVPage(TechDraw::DrawView* dView)
    - obsolete code deleted by PR#11103

4) bool GeometryMatcher::compareBSplines(TopoDS_Edge &edge1, TopoDS_Edge &edge2)
    - fixed by Werner PR #11095

5) void QGIViewPart::highlightMoved(QGIHighlight* highlight, QPointF newPos)
    - fixed by Werner PR #11096

12) int DrawUtil::getIndexFromName(const std::string& geomName)
    - missing '=' in DrawViewPart::getVertex() fixed by PR#11103

wwmayer, 2023-10-19:
Remaining issues:
3. `void Document::handleChildren3D(ViewProvider* viewProvider, bool deleting)`
7. `void SectionCut::startCutting(bool isInitial)` / class must be refactored anyway
13. `inline uint32 readUint32 ( istream &is )`

luzpaz, 2023-10-26:
Hey @sasobadovinac do you mind triggering a new coverity scan ? TIA!

sasobadovinac, 2023-10-26:
I was thinking to run all of them (Coverity, PVS-Studio and CodeQL) more at the end of the year, but if @wwmayer thinks it would make sense to do it now I can maybe do it next week...

wwmayer, 2023-10-26:
I don't mind when running it the next time. Running it next week or at the end of the year is fine for me.

wwmayer, 2023-10-26:
I plan to add the core system, the Part and Part design to the pre-commit stuff which will do quite a  lot of code-formatting.

The other thing is to run the clang-tidy checkers over the whole code base which ideally finds a lot of coding flaws that should be fixed before running  Coverity & co.

Then it's probably the best to do it in this order:
1. Code formatting with pre-commit
2. Run clang-tidy and fix the reported issues
3. Run Coverity

luzpaz, 2023-10-27:
An aside. should we add a section in the developer handbook about coverity/pvs-studio etc... when to run them and how to trigger new builds etc.. ?

sasobadovinac, 2023-10-27:
It would have to be setup, at the moment I am doing it all manually...

chennes, 2023-10-27:
> I plan to add the core system, the Part and Part design to the pre-commit stuff which will do quite a lot of code-formatting.

Do you have a timeline in mind for this?

wwmayer, 2023-10-27:
> Do you have a timeline in mind for this?

The formatting I would like to get done in the next couple of weeks. Fixing the issues discovered by clang-tidy will be a huge amount of work and I am not sure if I get this done until the end of the year.

chennes, 2023-10-27:
> Fixing the issues discovered by clang-tidy will be a huge amount of work and I am not sure if I get this done until the end of the year.

Yes, for sure! I'm sure you can get some help with this👍.

luzpaz, 2023-11-30:
What's left here ? (asking for housekeeping reasons)

wwmayer, 2023-11-30:
Point 3. and 13. For the former I haven't really understood what the code checker is trying to tell me and the latter is more of a theoretic issue.

luzpaz, 2024-08-25:
Added to post-1.0 list so this ticket doesn't get buried

chennes, 2025-06-08:
No: at this point it is too general. If the two issues @wwmayer references above are still unresolved, they should get extracted to their own GitHub issues. Otherwise they will just get lost again. These "omnibus" issues are hard for anyone to work on once a significant fraction of the sub-issues are resolved.