FreeCAD/FreeCAD #11098: [Problem] Background color on formula editor options is not readable

sdifru, 2023-10-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The background color on formula editor options is not readable. I can't see the options. On the **preference packs** I choose light theme and on the **style sheet** I use light-blue.

[image]

### Full version info

```shell
[code]
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: makepkg
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.11.5, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FreeCAD_assembly3 0.12.0
  * FreeCAD_FastenersWB 0.4.66
  * FreeCAD_SheetMetal 0.3.1
  * Lattice2 1.0.0
  * freecad.gears 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2023-10-20:
Did you do a reboot?
Some colors need a reboot?

sdifru, 2023-10-21:
> Did you do a reboot?

Yes.

I've change it and used Classic theme with light-blue. The black background color still exists. The previous freecad version doesn't have this.

I use sway compositor, not KDE or GNOME if it helps.

MisterMakerNL, 2023-10-21:
Hmm normally on linux this would happen if the background color isn't set, but on windows this has an gray background.
Which means it is set.
I also tested it with Fusion style which is often used in Linux. And I had the same results.
Could you run this macro and post the output from console?
```
class Qtprintstyles:
    from PySide import QtWidgets
    print(QtWidgets.QStyleFactory.keys())

```
This is how light looks on windows:
[image]

sdifru, 2023-10-22:
I use the beta stylesheets you mentioned https://github.com/MisterMakerNL/Freecad-Built-in-themes-beta and it works! Thanks mate @MisterMakerNL!

closed 2023-10-22 by sdifru