FreeCAD/FreeCAD #11102: [Problem] Main window doesn't remember where it was after restart

macdroid53, 2023-10-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Over the last few days I've noticed that the main window does not remember where is was when the program was shut down.

Curiously, the the Properties panel and the Task panel do remember (I typically keep them on a separate monitor).

When FC is started again, all three are displayed on the on the monitor where the Property/Task panels were located.

This have never been the case in the past and all remembered their previous position.

### Full version info

```shell
Previous version where found:
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34697 (Git)
Build type: None
Branch: main
Hash: cf9b342
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods:

    Silk 0.1.5
    Defeaturing 1.2.0
    Curves 0.6.13
    CurvedShapes 1.0.4
    DynamicData 2.53.0
    lattice2 1.0.0
    sheetmetal 0.3.0
    Manipulator 1.5.0
    kicadStepUpMod 10.19.5
    QuickMeasure 2022.10.28
    fasteners 0.4.66
    Help 1.0.3
    Assembly
    ProDarkThemePreferencePack 1.0.0
    A2plus 0.4.60n
    freecad.gears 1.0.0
    ExtremeProDark 2.9.0
    Assembly4 0.50.4
    [/code]
Latest version where found:
[code]
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34709 (Git)
Build type: Release
Branch: main
Hash: 7eda70c737625383d358871672d60a229e5e21b6
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * MnesarcoUtils 0.2.2
  * ZolkoBase 0.0.1
  * Pyramids-and-Polyhedrons
  * lattice2 1.0.0
  * Curves 0.6.13
  * Nodes 0.1.36
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.0
  * Assembly
  * Assembly4 0.50.4
  * freecad.gears 1.0.0
  * sheetmetal 0.3.1
  * Silk 0.1.5
  * kicadStepUpMod 10.20.3
  * A2plus 0.4.60n
  * DynamicData 2.53.0
  * workfeature-macro
  * fasteners 0.4.66
  * ExtremeProDark 2.9.0
  * Defeaturing 1.2.1
  * CurvedShapes 1.0.4
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

macdroid53, 2023-10-19:
It was not a problem in 0.21.x

wwmayer, 2023-10-19:
It's a regression caused by this change: https://github.com/FreeCAD/FreeCAD/commit/bf9584a90ba90984b8c3fb804025acba11795d2b#diff-a858c8d2d1dce20a3b475fd7013ed3348e86e11528b167843c72d304f4de12c0L1590

macdroid53, 2023-10-20:
@luzpaz   Same bad behavior in 34725 on Manjaro & Debian.

luzpaz, 2023-10-20:
Did the overlay code merge break this  ?

wwmayer, 2023-10-20:
> Over the last few days I've noticed that the main window does not remember where is was when the program was shut down.

What exactly does this mean? Is it on the wrong position or is on the wrong screen (in case you have two screens)?

macdroid53, 2023-10-20:
Both.
Before shutdown:
[image]

After shutdown an restart.
[image]

It used to come up as shown in the first image every time.

wwmayer, 2023-10-20:
I guess you are able to build FreeCAD on your own? If yes, then try these commits:
37e9aa9f9 -- the last commit before the PR 7888 was merged
bf9584a90 -- the part of PR 7888 that modified most of the main window
4eee90f6d -- fix Qt6 build after PR 7888 was merged

macdroid53, 2023-10-21:
> I guess you are able to build FreeCAD on your own? If yes, then try these commits: [37e9aa9](https://github.com/FreeCAD/FreeCAD/commit/37e9aa9f94e729b612f2cb3e8778ff73b6d3f00f) -- the last commit before the PR 7888 was merged [bf9584a](https://github.com/FreeCAD/FreeCAD/commit/bf9584a90ba90984b8c3fb804025acba11795d2b) -- the part of PR 7888 that modified most of the main window [4eee90f](https://github.com/FreeCAD/FreeCAD/commit/4eee90f6d40de3b8709b67527a4c0f7457ad8bae) -- fix Qt6 build after PR 7888 was merged

I am (I think) aware of how to build a particular PR. But, not a particular commit.

wwmayer, 2023-10-21:
If you have cloned a git repo on your system then you can e.g. do: `git checkout 37e9aa9f9` If you are done then you do `git checkout main`

The second commit contains some logic where the main window is moved to the screen where the splash screen was shown and this could be the culprit. Hence my question: with any version of FreeCAD when you start it does it show the splash screen on the left side?

macdroid53, 2023-10-21:
@wwmayer  Ok, my notes on building PR's referred to PR numbers only, so I didn't know I could just specify the commit # as well. I will try later this afternoon.

Side note: Today's main build now has regressed to the "if the panel is not docked, the panel does not give up focus when FC looses focus. The main window will fall behind, but the un-docked panels don't.

wwmayer, 2023-10-21:
If you want to switch back to the old docking behaviour then go to the preferences and disable "Activate overlay handling" on the General page.

macdroid53, 2023-10-21:
> If you want to switch back to the old docking behaviour then go to the preferences and disable "Activate overlay handling" on the General page.

Is this in reference to focus handling? The focus issue I referred to was around and has regressed a few times long before the overlay stuff.

wwmayer, 2023-10-21:
Which focus handling? The check box is about to disable the overlay stuff as it's not stable enough yet.

macdroid53, 2023-10-21:
> Which focus handling? The check box is about to disable the overlay stuff as it's not stable enough yet.

That's why I was asking. (See my side note above.) I was just noting behavior that has regressed in non docked panel previously.

macdroid53, 2023-10-22:
@wwmayer I built 37e9aa9f9 and it does remember the last location of the main window and the un-docked panels.

Hopefully I'll get to try the other two later today.

macdroid53, 2023-10-22:
4eee90f does not remember main window placement.

wwmayer, 2023-10-22:
OK, thanks. Then we know the regression started with bf9584a

macdroid53, 2023-10-31:
Does this regression have any traction? (I see it is now labeled help wanted. I can't offer coding help, but can test things.)

wwmayer, 2023-10-31:
> Does this regression have any traction?

We know the commit that caused the regression but since it's a fairly big change it's hard to say what exactly is the problem.

When you start FreeCAD you see a splash screen. On which side do you see it? Left or right?

macdroid53, 2023-11-02:
> > Does this regression have any traction?
> 
> We know the commit that caused the regression but since it's a fairly big change it's hard to say what exactly is the problem.
> 
> When you start FreeCAD you see a splash screen. On which side do you see it? Left or right?

The splash shows up on the screen where FC was when shutdown (in this case the right monitor of a 2 monitor setup). And, the main window shows up on the left monitor along with un-docked Properties and Task. (which is where they were when FC shutdown.

macdroid53, 2023-11-03:
@wwmayer 

`>>> QtWidgets.QApplication.primaryScreen().availableGeometry()
PySide2.QtCore.QRect(0, 0, 1920, 1080)
>>> QtWidgets.QApplication.desktop().availableGeometry(-1)
PySide2.QtCore.QRect(0, 0, 1920, 1080)

<input>:1: DeprecationWarning: QDesktopWidget.availableGeometry(int screen) const is deprecated
>>> QtWidgets.QApplication.desktop().availableGeometry(0)
PySide2.QtCore.QRect(0, 0, 1920, 1080)

<input>:1: DeprecationWarning: QDesktopWidget.availableGeometry(int screen) const is deprecated
>>> QtWidgets.QApplication.desktop().availableGeometry(1)
PySide2.QtCore.QRect(1920, 0, 1920, 1080)

<input>:1: DeprecationWarning: QDesktopWidget.availableGeometry(int screen) const is deprecated
>>> 
>>> for scr in QtWidgets.QApplication.screens():
...     print (scr.availableGeometry())
... 
PySide2.QtCore.QRect(0, 0, 1920, 1080)
PySide2.QtCore.QRect(1920, 0, 1920, 1080)
`

Switching splash screen off, same behavior.

wwmayer, 2023-11-03:
Thanks for testing.

So, according to the test results the size of both screens is 1920x1080 and the first screen starts at 0x0 and the second at 1920x0.

In the past there was no specific handling of 1st or 2nd screen but `QApplication::primaryScreen()->availableGeometry()` was used and this has worked correctly.

Now there is an explicit handling of 1st and 2nd monitor -- the screen that shows the splash screen is used now.

> Switching splash screen off, same behavior.

If the splash screen is switched off then the primary screen like in the past is used. Unfortunately, the changes caused some more damages because it has removed the config settings from FreeCAD.conf.
So, in order to test this properly again do the following:
1. Use an older FreeCAD version, put the main window on the right screen and close the application. Important: this version must link the same Qt version as your v0.22 build of FreeCAD as otherwise the required config settings won't be restored
2. Restart the old FreeCAD version to make sure it works as expected
3. For the v0.22 build make sure that the splash screen is off and start it
Is it still on the wrong side?

wwmayer, 2023-11-05:
https://github.com/FreeCAD/FreeCAD/pull/11275 is supposed to fix the issue. Are you able to test the PR?

macdroid53, 2023-11-05:
> #11275 is supposed to fix the issue. Are you able to test the PR?

I should be able to later today. But, (I am really limited in the ways of PR's and git.) How do I get the number to check out from the the PR? Is it just: "git checkout 11275" ?

wwmayer, 2023-11-05:
An easy way is to get the diff file https://patch-diff.githubusercontent.com/raw/FreeCAD/FreeCAD/pull/11275.diff and save it to your FreeCAD folder.
Then apply the path with `git apply 11275.diff`. After finishing the tests you can revert the changes with `git checkout src\Gui\MainWindow.cpp`

macdroid53, 2023-11-05:
@wwmayer  So, I do a normal git clone of the source. Download that diff file, then apply. Then build as usual?

macdroid53, 2023-11-05:
@wwmayer yes, the main window and undocked panels retain their last position on restart.

wwmayer, 2023-11-05:
OK, thanks for testing.

macdroid53, 2023-11-08:
Just curious, this has apparently not made it to a commit. Or am I missing something?

wwmayer, 2023-11-08:
There are some concerns that restoring the main window on a multi-monitor setup on Windows won't work with that PR. But I don't know how the current behaviour is.

Now looking at the Qt API I found a method that exists since version 4.2 to save and restore the geometry of a widget. This code explicitly handles multiple monitors of a system and may work on all OS.

I can prepare a 2nd PR that uses these Qt functions.

maxwxyz, 2023-11-10:
Does it handle https://github.com/FreeCAD/FreeCAD/issues/11142 as I am still not able to build it myself?
- open on Windows with three monitors
- move to third monitor and close
- switch laptop to a two monitor setup
- is main window reachable?

Currently some dialog windows are out of my reach at home (preferences or property editor) with no way of getting them from the missing display to a visible space.

wwmayer, 2023-11-10:
https://github.com/FreeCAD/FreeCAD/pull/11275#issuecomment-1805231598

closed 2023-11-14 by adrianinsaval