FreeCAD/FreeCAD #11105: Released disk image for arm mac requires rosetta

jmgurney, 2023-10-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The dmg that claims to support M1/M2 arm Macs does not work w/o Rosetta.  It pops open a dialog telling me to install rosetta.

### Full version info

```shell
I cannot use about box since FreeCAD refuses to run, but I have tried `FreeCAD-0.21.0-mac-arm64.dmg`, `FreeCAD-0.21.1-mac-arm64.dmg` and `FreeCAD-0.21-RC1-mac-arm64.dmg`.
```

### Subproject(s) affected?

None

### Anything else?

Machine is a 2021 14-inch MBP w/ an M1 Pro running macOS 14.0 (23A344).  It does not have rosetta installed, partly to discover programs which claim to support ARM, but actually do not.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jmgurney, 2023-10-19:
I also just attempted w/ `FreeCAD_weekly-builds-34667-2023-10-15-conda-macOS-arm64-py310.dmg` and it also prompts to require Rosetta.

klokik, 2023-10-20:
Workaround: launching executable directly works `FreeCAD_0.21.1_arm64 % ./FreeCAD.app/Contents/MacOS/FreeCAD`

felipebalbi, 2023-12-31:
I believe the issue is because of the shell script runner. Perhaps the environment variables set by the script should be moved to `Info.plist` and the binary should be placed in `MacOS/FreeCad` instead.

dansharkey, 2024-08-08:
> I believe the issue is because of the shell script runner. Perhaps the environment variables set by the script should be moved to `Info.plist` and the binary should be placed in `MacOS/FreeCAD` instead.

I can confirm this is the issue. The "executable" in /Applications/FreeCAD.app/Contents/MacOS is actually a bash script that sets some environment variables and then calls the true executable which is in /Applications/FreeCAD.app/Contents/Resources/bin/freecad. Since the MacOS directory doesn't contain the arm executable, then Mac OS incorrectly assumes that the package requires Rosetta 2.

By moving the true exe into the MacOS/ directory it is now showing as "Application (Apple silicon)" when FreeCAD.app is right-clicked and Get Info is selected, and the Rosetta 2 requirement disappears.

The problem now is that since the exe has moved then all of the dynamically linked libraries can no longer be found, which I guess is what some of the environment variables (LD_LIBRARY_PATH) are attempting to do in the original bash script. Looking at [FreeCAD-Bundle/blob/main/conda/scripts/fix_macos_lib_paths.py](https://github.com/FreeCAD/FreeCAD-Bundle/blob/main/conda/scripts/fix_macos_lib_paths.py) and FreeCAD-Bundle/blob/main/conda/osx/APP/FreeCAD.app/Contents/MacOS/FreeCAD seems to indicate that @adrianinsaval and @looooo have some experience in this area?

After moving the executable to MacOS, I get the error
```
dyld[8043]: Library not loaded: @rpath/libFreeCADGui.dylib
  Referenced from: <51373041-D1BF-37DE-8F04-E1B6A0CB9ACE> /Applications/FreeCAD.app/Contents/MacOS/freecad
  Reason: tried: '/Applications/FreeCAD.app/Contents/MacOS/../lib/libFreeCADGui.dylib' (no such file), '/Applications/FreeCAD.app/Contents/MacOS/../lib/libFreeCADGui.dylib' (no such file), '/usr/lib/libFreeCADGui.dylib' (no such file, not in dyld cache)
zsh: abort      /Applications/FreeCAD.app/Contents/MacOS/freecad
```
which I can work around by also moving all of the contents of Resources up into the top level of FreeCAD.app (this lets it run by double-clicking, with no Rosetta warning, after a bit of `xattr -dr com.apple.quarantine /Applications/FreeCAD.app` trickery)

I'm afraid that's as far as I've been able to get - actually fixing the issue may be as simple as running a tool like `install_name_tool` after moving the exe, and adding the remaining environment vars to Info.plist as @felipebalbi suggests. Or it may be best to tweak the aforementioned `fix_macos_lib_paths.py` script, or even resolve the CMake scripts so the correct dynamic library paths are created during the build.

oursland, 2024-08-08:
I am somewhat familiar with the packaging system, although it is indeed maintained by @adrianinsaval and @looooo who are far more familiar.  I believe until now, none of us knew the reason for requiring Rosetta 2, but with this investigation there's a path forward.

adrianinsaval, 2024-08-13:
one can just do a symlink instead of copying the executable, the problem with this is that we can't properly set the env variables without the bash script. I tried now adding a value to Info.plist that should tell mac what architecture the bundle supports so hopefully it won't try to run with rosetta anymore. Could you please test the latest weekly?