FreeCAD/FreeCAD #11107: [Problem]

jffmichi, 2023-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Profile mixes climb and conventional milling when processing open edges. The edges are sorted so that the travel moves are minimized without respecting the selected cutting direction. A pull request that fixes this issue is on the way.
[image]
[profile_open_edge_direction.zip](https://github.com/FreeCAD/FreeCAD/files/13048624/profile_open_edge_direction.zip)

### Full version info

```shell
[code]
OS: Arch Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34715 (Git)
Build type: None
Branch: main
Hash: 9defdaf0a66bed1fdf9ca08b93080d052bfa05b8
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * better-tool-library 0.9.9
  * dodo 1.0.0
  * sheetmetal 0.3.1
  * DynamicData 2.53.0
  * fcgear 1.0.0
  * fasteners 0.4.66
  * fcgear.backup1663629032.4145591 1.0.0 (Disabled)
  * Assembly4 0.50.4
  * lattice2 1.0.0
[/code]
```

### Subproject(s) affected?

Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-20:
@LarryWoestman do you mind responding to this Path ticket? TIA!

LarryWoestman, 2023-10-23:
I'm sorry, but I can't figure out if the original code is correct or what it is doing in many areas, so I have no confidence that the new code is an appropriate change.

There is a lot of background information about the internal data structures that I don't know and can't figure out from the available documentation.

closed 2023-10-24 by jffmichi