FreeCAD/FreeCAD #11109: Unable to change values in FEM temperature constraint - M1 MacOS 0.21.1

crowdis8, 2023-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Opening up FEM workbench produces same error to https://github.com/FreeCAD/FreeCAD/issues/10285

Using FreeCAD-0.21.1-mac-intel_x86-unsigned.dmg install on a M1 Mac Venture 13.6

When adding FEM Constraint Temperature, the attribute is added to the Analysis tree and I can open the menu but when I change any of the attributes (Temperature or Concentrated Heat Flux) and hitting Ok the numbers are not saved. I change them they just restore back to default 300K and 0.00 mW when I reopen the menu. It is obvious that the numbers have not been registered because when I run the Solver there is no temperature variation (max vs min), meaning it has not registered the heat flux.

Going back to version 0.20.2 the problem is not present.

[image]
[image]
[image]

### Full version info

```shell
[code]
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

FEM

### Anything else?

I initially installed the 0.21.1 Arm64 image, but opening up FEM workbench produces same error to https://github.com/FreeCAD/FreeCAD/issues/10285

Error:
dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical'

So I tried the solution and installed the 0.21.1 x86 unsigned image, which resolves the issue with the FEM workbench not working. However there is still the issue as described above.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-20:
@FEA-eng :point_up:

marioalexis84, 2023-10-20:
Confirmed with c85bc2a75534afd363e40c8569d9922971400ab1

PR #11126 (2023-10-23): Fem: Connect constraint temperature spinboxes to properties and task dialog rework - fixes #11109
This fixes #11109.
Change the radio buttons for one combo box with the `ConstraintType` values.
Depending on the value of this property, the `Temperature` spinbox or the `Concentrated heat flux` spinbox becomes visible.
commit c728676d (2023-10-23): Fem: Connect constraint temperature spinboxes to properties and task dialog rework - fixes #11109
commit ee9b1b9e (2023-10-23): Merge pull request #11126 from marioalexis84/Fem-task_temperature

Fem: Connect constraint temperature spinboxes to properties and task dialog rework - fixes #11109
commit e11d5dff (2023-11-10): Fem: Connect constraint temperature spinboxes to properties and task dialog rework - fixes #11109
closed 2023-10-23 by sliptonic