FreeCAD/FreeCAD #11111: [Problem] Pad / Extrude "up to Face", "to First", "to Last" not working as expected in Part Design

maxwxyz, 2023-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Part Design Pad Feature Length Attributes "up to face", "to last" and "to first" are not working as expected or like in any other CAD applications.
It seems that currently the center of the profile/sketch is projected in the normal direction of the sketch and this would result in a static dimension for the extrude length.

This can lead to problems when the center of the profile is extended and does not find a face because of a hole or a curving face as Shalmeneser pointed out [here](https://forum.freecad.org/viewtopic.php?t=81740): 
[image]

Also, the profile is extended in the pad direction and ends with a parallel face to the profile/sketch plane. 
Expected would be a behavior, where the extrude of the pad feature seamlessly join with the curved face it should be padded to like in this example of other CAD programs:
[image]

Currently in FreeCAD a pad "to last" produced the following wrong result, when the outer shell face is selected:
[image]
A workaround of a manual length does not work with thin and curved objects as the planar extrude would protrude or stick out of the other side and must be trimmed manually (painful on casted or molded parts).
In the example above, the pad operation with length set to "to first" or "to face" does not work at all, as the distance currently determined by the pad feature is too short to join the extruded profile with the existing geometry which would result in a multibody (error: "resulting shape is empty").

### Full version info

```shell
0.21
0.22dev
```

### Subproject(s) affected?

PartDesign

### Anything else?

https://forum.freecad.org/viewtopic.php?t=81740

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2023-10-20:
Related to #7189 and it's partial fix #9743.

adrianinsaval, 2023-10-20:
> Currently in FreeCAD a pad "to last" produced the following wrong result, when the outer shell face is selected:

can you provide an example file for this? I cannot reproduce this, and please always provide full version information
[image]

adrianinsaval, 2023-10-20:
you are right that the centroid of the profile is projected in the pad direction, but this is only used to find which face should be used as limit of the pad, then that faced is passed on to occt to use as limit, we do not pass just a length value calculated from this intersection.

maxwxyz, 2023-10-20:
Please try to edit the length dimension in the attached file to anything like to face, to last or to first. None will work in 0.22dev.
[extrude.zip](https://github.com/FreeCAD/FreeCAD/files/13052916/extrude.zip)

FlachyJoe, 2023-10-20:
@maxwxyz UpToFace is working as expected on my side. Only UpToFirst and UpToLast fail because of the looking for face along the CoG normal.
As I point before it's an OCC limitation and I can't see any workaround.

@adrianinsaval :+1: I never see the Pad stops perpendicularly at the length of the intersection.

adrianinsaval, 2023-10-20:
> Please try to edit the length dimension in the attached file to anything like to face, to last or to first. None will work in 0.22dev. [extrude.zip](https://github.com/FreeCAD/FreeCAD/files/13052916/extrude.zip)

here the problem seems to be that the profile is outside of the bounds of the limiting face, and we currently have no way of selecting more than one face as limiting surface (and I'm not sure if occt supports this)

maxwxyz, 2023-10-20:
in the attached example it is working with three faces (angled to each other and the second with a fillet to a third face) 
[image]
[multiplefaces.zip](https://github.com/FreeCAD/FreeCAD/files/13054172/multiplefaces.zip)

maxwxyz, 2023-10-20:
> @maxwxyz UpToFace is working as expected on my side. Only UpToFirst and UpToLast fail because of the looking for face along the CoG normal. As I point before it's an OCC limitation and I can't see any workaround.

Up to face is not working on either available faces for me in current 0.22 weekly build and stable 0.21.1: 
[image]
How and what was your result?

luzpaz, 2023-10-20:
Folks, please use full [About](https://wiki.freecad.org/About) info. It saves a lot of time and confusion.

maxwxyz, 2023-10-20:
> Up to face is not working on either available faces for me in current 0.22 weekly build and stable 0.21.1: [image] How and what was your result?

see below, not working and also in current stable.
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.24.5
  * Curves 0.6.13
  * OpenDark 2023.10.9
  * sheetmetal 0.3.1
[/code]

FlachyJoe, 2023-10-20:
@maxwxyz please provide the failing file.

maxwxyz, 2023-10-20:
It is right there: https://github.com/FreeCAD/FreeCAD/issues/11111#issuecomment-1772432318

adrianinsaval, 2023-10-20:
> in the attached example it is working with three faces (angled to each other and the second with a fillet to a third face) 

in this case the center plane face is used as bound when doing up to first and plane faces are extended as infinite planes for the operations, by chance that gives the expected result here. If you do up to face and select the other plane face you'll see that is used as an infinite plane as the limit:
[image]
that doesn't work if you select the curved face though:
[image]

the problem in these cases is that round faces cannot be reliably extended to cover the full profile like we can for plane faces

maxwxyz, 2023-10-20:
Attached is another one failing in stable FC. Try changing the last Pad Feature to either "to first", "to last" or "to face". In all cases it is either an error or protruding through the object:
[example2.zip](https://github.com/FreeCAD/FreeCAD/files/13057597/example2.zip)
[image]
[image]
[image]
and here I have tried all different faces:
[image]
[image]
[image]

The right result should look in all cases like this:
[image]

In this case I have just removed the protruding material, but in complex parts (castings/molded parts) no chance to always remove all excess material for each pad operation.

[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.5
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * Defeaturing 1.2.1
  * fasteners 0.4.66
  * freecad.gears 1.0.0
  * freecad_metal_workbench
  * OpenDark 2023.10.9
  * sheetmetal 0.3.1
  * Silk 0.1.5
[/code]

adrianinsaval, 2023-10-20:
to solve this first we have to check if occt supports passing multiple surfaces as limit for prism generation. If yes, then we need to work on improving the "upToFace" selection mechanism (which seriously needs improvement anyways) to allow selecting multiple faces and lastly we need to change the up to last/first algorithms, perhaps we can project lines from each vertex in the profile instead of just from the centroid.

maxwxyz, 2023-10-20:
Attached another simple example where every operation is generating an error.

[example3.zip](https://github.com/FreeCAD/FreeCAD/files/13057737/example3.zip)

Try adjusting the distance below either to first, to last or to face.
[image]

FlachyJoe, 2023-10-21:
If the projection is partially out of the selected face it fails and it's a normal behavior. Where should the prism stop?

In _example3_ you can't select a face which totally constrain the prism end. At the opposite this is valid as the projection is contained by the selected face:
[image]

@maxwxyz All the problems you reports are related to the impossibility to select multiple faces and by the GoC projection to find the limit.
The multiple faces selection need a further dig into OCC sources. I plan it.

Proposed workaround to Pad to multiple faces:

1. Split the sketch in a way each part is projected in a unique face.
2. Pad each part to the subsequent face.

maxwxyz, 2023-10-21:
If there is a solution this would not only help the Pad feature but maybe also hole, pocket, loft and especially to find the "end" for https://github.com/FreeCAD/FreeCAD/issues/10538

Is it possible to somehow project the geometry of the profile in the direction of the extrude onto the body / other remaining geometry and loft to between those two to create a workaround (or maybe as an idea for the pad feature)?

FlachyJoe, 2023-10-21:
About CoG projection the possibilities are:

- Improve OCC to use vertexes instead of CoG -> an OCC issue should be opened
- Create our own algorithm to make the prism -> low level shape manipulation is not in the FreeCAD aims and it require sufficient acknowledgment.
- Imagine a workaround to use the OCC implementation as is -> only needs time and tests.

About multiple faces as end:

- Check for OCC capabilities and implement the faces selection
- Automatize the workaround proposed above and implement the faces selection

Note: this report confuses two problems, the already known Issue with CoG projection and the Feature Request for multiple faces selection.
IMHO this should be closed in favor of #7189 reopening (only partially fixed) and a clear new FR.

maxwxyz, 2023-10-21:
I cannot reopen https://github.com/FreeCAD/FreeCAD/issues/7189, can you describe the clear new FR with the differences which should already be fixed?

FlachyJoe, 2023-10-21:
@maxwxyz FR done with #11135

@AjinkyaDahale can reopen #7189

adrianinsaval, 2023-10-22:
> Improve OCC to use vertexes instead of CoG -> an OCC issue should be opened

I could be wrong but I think this is a FreeCAD problem not an OCC problem

adrianinsaval, 2023-10-22:
> @AjinkyaDahale can reopen #7189

I think it's better to create a new issue for the cog problem, that one is really about faces that are intersected twice

FlachyJoe, 2023-10-22:
> I could be wrong but I think this is a FreeCAD problem not an OCC problem

The concerned OCC code was linked yet: https://github.com/FreeCAD/FreeCAD/pull/9743#issuecomment-1583202179

luzpaz, 2024-12-09:
@FlachyJoe do you mind checking what's left to do here?

FlachyJoe, 2024-12-10:
We still can't pad to a closed cylinder internal because the prism stops at the first frontier but I think all the other problems was solved.
[image]

luzpaz, 2024-12-11:
> We still can't pad a hollow profile to a closed cylinder internal because the prism stops at the first frontier  

Should we close this and open a separate ticket for that ?

FlachyJoe, 2024-12-11:
Yes, a clean new issue is a good option.

luzpaz, 2024-12-11:
Closing as complete. Remaining issue has been split into a new ticket 
* #18436

closed 2024-12-11 by luzpaz