FreeCAD/FreeCAD #11112: [Problem] TechDraw: Feature for GD&T or Improvement of the Balloon Workaround

maxwxyz, 2023-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is no explicit feature for GD&T (https://en.wikipedia.org/wiki/Geometric_dimensioning_and_tolerancing) in TechDraw.
I know that it would be ideal to implement in the 3D environment and derive the drawing in TechDraw later but there is currently no active or working GD&T Workbench in FreeCAD. https://github.com/5axes/FreeCAD-GDT

As ToniTen mentioned [here ](https://forum.freecad.org/viewtopic.php?t=68639) the current workaround to hack GD&T into rectangular balloons is not ideal and produces results which need manual adjustments of the final .svg drawing in external applications.

If a dedicated GD&T feature for ISO GPS is neglected in FreeCAD, we should at least have a better workaround to be able to use FreeCAD in professional environments:

- Icons and text in rectangular balloons are currently not centeres
- No feature to create multi-line balloons 
- Remove excessive whitespace inside the rectangles
- Make the first rectangle always a square and the same size when using  GD&T symbols in it
- Make the leader line connected to the balloon features
- Include all operators of ISO GPS e.g.:
[image]

### Full version info

```shell
0.21
```

### Subproject(s) affected?

Techdraw

### Anything else?

https://forum.freecad.org/viewtopic.php?f=10&t=22072

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-12-06:
Most icons are in ISO 1101

sliptonic, 2024-04-22:
@maxwxyz  there are a handful of issues related to GD&T.  Will you create a project to group them?  Then we can start tracking them together and look at getting some FPA grant money assigned.

maxwxyz, 2024-04-22:
@sliptonic yes, could you create a project, we need an organization owner for this. If possible, please also one for UI stuff.

sliptonic, 2024-04-22:
Done.  I've invited you to both as an admin

https://github.com/orgs/FreeCAD/projects/23/views/1
https://github.com/orgs/FreeCAD/projects/24/views/1

maxwxyz, 2024-04-22:
@sliptonic thanks, will add matching issues and create new ones.
Is it meant for TechDraw only or also in the 3D view? Best way would be support in the 3D view as most applications do to add tolerances to datums on the model and derive the drawing from it. Faster could be the implementation in 2D only I guess.

There was an addon for the 3d implementation but it is not maintained: https://github.com/5axes/FreeCAD-GDT

sliptonic, 2024-04-22:
Yeah,  GD&T is a bigger topic.  It certainly affects TechDraw but also the 3D window and the Step export facility.  

There's a fork of the addon that works and isn't in the addon manager.   

Basically the FreeCAD community needs to have a bigger discussion about GD&T and how we support it.

maxwxyz, 2024-04-22:
I guess it depends on the resources. 3D support + step would be the go to for professional applications but easier could be the hack with the balloon feature.

lucas-bueno777, 2024-04-22:
Hi! I am not sure if this is the best place to comment about this, but I like the idea of adding GD&T in the 3D view, such as Functional tolerancing and Annotation (FTA) for CATIA. It is widely used in the industry - but probably is very hard to implement.

Thanks for the amazing work.

maxwxyz, 2024-04-24:
@sliptonic there are not more issues directly related to GD&T. I could go through each ISO GPS standard and create separate issues of what is needed to fully implement this. 
But I guess this is a lot of work and will need some time as it's not just a list of what's missing but it has to be grouped and ideally proposed as a functioning UI concept.

I could also try to create the issues so they are for the 3D implementation and the drawing could be derived later, as GD&T should be done professionally. So feature requests for a GD&T WB and TechDraw could grab this info for the drawing.
If this could be implemented in 3D via a new GD&T WB there could be also the possibility for future WBs to build on top using the information for tolerance analyzes / variance analyzes and so on.

Is there any other person who is familiar with GD&T who could support?

This is just the base of standards I've grabbed from an book:
[image]

sliptonic, 2024-04-24:
Even the assumption that there should be "a GD&T workbench" may be premature.  For all we know, GD&T functions rightly belong in suitable places throughout the application.  

I don't think we should go and create the whole raft of issues up front.  I'd prefer to have the issues be driven from user-facing problems.  We could hunt through the forum and see if users have been asking for GD&T functionality.  Then tap them to create specific issues in the tracker.

I don't know of any local experts in the field.  Issue #1 might be to identify a project manager to figure out what needs to be done. That person might come from the forum or from another source.  I'll go through my contact information to see if I know anyone.

pure-machinery, 2024-08-30:
From my side it is best to tackle 2D first, but this would require better logic for annotations or leader lines even before GD&T symbols are added. The symbols can just be used from a .ttf file.

If 3D is tackled first then I would say there should be an ability to pull it into 2D from the model. This would be awesome but I find it hard to believe FC will do something like this. 

Creo has pulling dimensions and annotations from features into the drawing for example. This makes creating drawings very easy and it makes it hard to forget if you've added a specific dimension.

cooper-david-a, 2024-10-17:
The problem I've always had with GD&T in 3D models is that the vendor with whom you are communicating needs to be able to open the 3D file.

The latest .step files are the only neutral format that I am aware of that will preserve the annotations and I have not been very impressed with the result so far.  I **don't** think the 3D pdf views produced by MBD features provide much actual utility.

Unless you do all of your manufacturing in house or are a large company you have to work with what your vendor uses.  Typically in my case this means providing a 3D model in a CAD neutral format for them to use in their CAM software and a 2D drawing to handle all of the tolerances and notes.

Once you've arrived at the point where 3D annotations are not useful in getting parts produced, there is no point in placing them in the 3D environment at all.

Honestly, the manual method of attaching GD&T symbols I saw here [here](https://youtu.be/7GwaIpJgp1Q?si=vAKHA9P0FGG64uD8) is adequate.  Maybe some minor tweaks like making the symbols look better and allowing feature control frames to be grouped with dimensions (so they move together).

I am actually most interested in a CAD software that will run on Linux.

maxwxyz, 2024-10-17:
IMO 3D attachment is crucial. Not only for exchanging in STEP format but also the ability to perform tolerance analysis in assemblies. For manufacturing and quality control (manually) the 2D drawing is sufficient. But there are measurement machines out there that can work with 3D models, grad the properties (product specifications) per surface / axis out of the 3D file and derive the right measurement program and compare the sample.
The attachment like annotations in 3D are only interesting for the human and to be able to assign tolerances and check them. The data should be stored as properties for the selected features.

cooper-david-a, 2024-10-18:
I'm not familiar enough with tolerance stackup or  measurement software to comment on those use cases.  I am curious whether there will not need to be interfaces with a bunch of proprietary file formats or if those software support some neutral format.

It looks like [AP242](http://www.ap242.org/welcome.html) is addressing the interoperability issues I had with MBD last time I looked into it.  If the latest systems are capable of importing the GD&T from a neutral format like .step AP242 in a usable way and not just the annotations, that changes things.

Until 3D data is supported broadly, (which might be happening now) I would just hate to loose the simple and elegant solution of updating GD&T feature control frames right there on the 2D drawing.

luzpaz, 2024-10-18:
Heads up: seems Ondsel is interested in implementing GD&T https://ondsel.com/blog/gd-and-t-as-core-feature/

lucas-bueno777, 2024-10-18:
[Here](https://www.wichita.edu/industry_and_defense/NIAR/Documents/Functional-Tolerancing-Annotations-R30.pdf) is a publicly available file that shows how CATIA handles GD&T, in case that helps in the brainstorming.

cooper-david-a, 2024-10-18:
> Heads up: seems Ondsel is interested in implementing GD&T https://ondsel.com/blog/gd-and-t-as-core-feature/

Yeah, I read that article.

For some background, I am a mechanical engineer and have been working in the aerospace industry for around 13 years now.  I've used Solid Works and SolidEdge extensively and have been recently using Fusion.

I played around with FreeCAD about 10years ago for a year or two.  I just started following it again as I am interested in getting away from Windows.

It could be that I am unusual but when I am developing geometry in CAD I am doing something fundamentally different then when I am making a 2D drawing to send to a manufacturer.  Developing the geometry can be a messy process where it makes sense to define geometry off of mating components or reference planes or other weird stuff that wouldn't be in a 2D drawing.  Making a 2D drawing is specifying a set of manufacturing requirements.  If you are able to do both at the same time, you are much better at this than I am.

I personally haven't found the ability to import dimensions from the 3D geometry very useful at all.  I always start with a blank view and add dimensions and tolerances in the 2D environment.

Progress is apparently being made toward a 3D only manufacturing workflow.  However, I think it is still a long way off for those of us who can't mandate what software our vendors use.

Mannshoch, 2024-10-19:
@cooper-david-a There are two concepts about messuring GD&T/ISO-GPS one is about following the design features and one is about, easuring in 3D after the design. Your Way of Working is mor about Measuring after the design, but that could also happen in 3D.
I know companies, that which have the rule, that the moment before a part switch from developing to the first release the part has to be newly created from the Ground. Benefit is, that while developing a part you have not to focus on a sustainable feature tree. A redesign is mostly done really fast and while doing you could design it according to needs for variants, and measure it for auto import into Drawings, and [add comment](https://help.solidworks.com/2025/english/solidworks/sldworks/hidd_comment_dlg.htm).

If you work that way you could also add tolerances and GD&T/ISO-GPS symbols.

I assume a combined possibility would be best. Measure it in sketch, and add gd&t there if useful and otherwise add gd&t from external. 

I think, influenced by Solidwork, there must be a possibility to see all GD&T/ISO-GPS symbols used in 3D and modify from there. Like removing/attaching a symbol to a sketch-dimension and/or add/remove a 3D-Measure to the body.

cooper-david-a, 2024-10-19:
@Mannshoch, Good points.  I may try the method you describe on my next few drawings.  I don't understand the advantages of such an approach at this point but I'll report back.  I'll also discuss this with some of my suppliers.  They may be further along than I am giving them credit for.

pure-machinery, 2024-10-30:
One thing I would like to add is that this will affect the snap lines implementation in Freecad. 

Currently we have a fixed width between dimensions but what happens if you have a tolerance frame (or multiple) beneath the dimension line? 

You could argue that the tolerance frames could be used when calculating the height of the dimension and just have the same offset between the last tolerance frame and the highest point of the dimension below.

pierreporte, 2025-04-29:
A **150 € reward** will be given to the contributor that writes a pull request, solving this issue, that gets merged. It will be paid by the FreeCAD Project Association. It is to be considered an incentive. Grant applications are welcome, but the total amount will exclude the reward.

franklincg, 2026-08-17:
Hi, I'm interested in working on this bounty. Could you confirm the current reward amount and whether the issue is still available for a new contributor to work on?

mohamed-khairy-swe, 2026-08-17:
@franklincg 
sorry but I'm currently working on this issue, but I encourage you to work on this bounty #8076 because it would be a really nice addition to FreeCAD, and because you are a new contributor please don't use AI , try to tackle the issue on your own and plan your approach on how you are going to solve it and read the codebase and really put that brain of yours to work.