FreeCAD/FreeCAD #11113: [Problem] Adjust Default Environment Light Position

maxwxyz, 2023-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Either let the user adjust the global position of the light / illumination of the environment or adjust the default position so that is not directly coming from the direction of the viewing plane / angle.

I use a lot of material and color variations for different bodies and faces and when they are glossy and face directly the view, they all appear white (reflection of the environment lightning). It would be beneficial to rotate the global light source to an angle that when drawing on standard planes, it is possible to distinguish different parts and faces.

Especially when doing sketches I cannot differentiate between different color faces because of this behavior, all just appear white.

As Syres mentioned, it is currently only possible to adjust the position with Python but that has to be done at every restart:
https://forum.freecad.org/viewtopic.php?p=146925#p146925

### Full version info

```shell
0.21
```

### Subproject(s) affected?

Core

### Anything else?

https://forum.freecad.org/viewtopic.php?p=676725
https://forum.freecad.org/viewtopic.php?p=146925

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-10-21:
Here is a demonstration. Simple model but you cannot distinguish the colors/faces or features you want to see, when viewing it directly from the normal direction of the faces. All appear white. 
You can take alook in the file and edit the first sketch: 
[environment_light_view-from-top.zip](https://github.com/FreeCAD/FreeCAD/files/13061634/environment_light_view-from-top.zip)

[image]

maxwxyz, 2023-10-22:
Thanks, great description: http://catiadoc.free.fr/online/basug_C2/basugbt1701.htm
If [this ](https://forum.freecad.org/viewtopic.php?f=10&t=9663&start=60#p394930)is implemented in the @realthunder fork, can this be implemented in main with a PR?

wwmayer, 2023-10-22:
https://github.com/FreeCAD/FreeCAD/pull/11146

xtemp09, 2023-10-25:
Has this issue been solved with #11146?

maxwxyz, 2023-10-25:
I hope, but cannot build FC myself so I'm waiting for the next weekly build.

maxwxyz, 2023-10-26:
Works flawlessly, thanks!

closed 2023-10-26 by maxwxyz