FreeCAD/FreeCAD #11118: Mesh intersection not working correctly for these two meshes

NulAsh, 2023-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Import these two files: ComponentM079.stl  and ComponentM083.stl [ComponentM0.zip](https://github.com/FreeCAD/FreeCAD/files/13053381/ComponentM0.zip)
2. Switch to Mesh Workbench
3. Select both imported meshes
4. From the main menu Meshes -> Boolean -> Intersection

Desired result: created mesh with the name "Intersection", containing actual intersection of M079 and M083 (two small objects)
Actual result: created mesh with the name "Intersection", containing one of the meshes M079 or M083 (depending of the order in which you select them - it will return the first selected mesh)
[image]

In Blender everything works fine.
[image]

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: makepkg
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.11.5, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
  * woodworking 0.21.30486
  * pcb 6.2023.0
  * Ship 2022.4.11
  * OpticsWorkbench 1.0.9
  * POV-Ray-Rendering
  * InventorLoader 1.4.0
  * AnimationFreeCAD 1.0.0
  * FEMbyGEN 1.2.0
  * Assembly4 0.50.4
  * sheetmetal 0.3.1
  * CurvedShapes 1.0.4
  * Assembly3 0.12.0
  * Manipulator 1.5.0
  * A2plus 0.4.60n
  * Curves 0.6.13
  * ArchTextures
  * AirPlaneDesign 0.4.0
  * Render 2023.8.18
  * Rocket 3.2.3
  * kicadStepUpMod 10.20.7
  * fasteners 0.4.67
  * ExplodedAssembly
  * BCFPlugin 1.0.0
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-10-20:
Mesh_Evaluation (https://wiki.freecad.org/Mesh_Evaluation) reports error for both meshes:
ComponentM079: 24 folds on surface
ComponentM083: 65 folds on surface

Roy-043, 2023-10-20:
Closing this.

closed 2023-10-20 by Roy-043