FreeCAD/FreeCAD #11120: [Problem] Network Manager Error

wwmayer, 2023-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Network Manager Error: __reply_finished not called by a Qt signal" error while trying to run the addon manager 

Traceback (most recent call last):
File "/tmp/.mount_FreeCAja2QAS/usr/Mod/AddonManager/addonmanager_connection_checker.py", line 80, in cancel_network_check
self.connection_checker.success.disconnect(self._check_succeeded)
RuntimeError: Failed to disconnect signal success()

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Addon Manager

### Anything else?

https://forum.freecad.org/viewtopic.php?t=82060

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-30:
Direct link to source: https://github.com/FreeCAD/FreeCAD/blob/d529193f751e212fa5e6f08374e2cb452ea82b3a/src/Mod/AddonManager/addonmanager_connection_checker.py#L80

luzpaz, 2023-11-30:
What's the solution here ?

PR #12126 (2024-01-27): Addon Manager: Don't fail during signal disconnect
Fixes #11120
commit f0a1e049 (2024-01-27): Addon Manager: Fix #11120
PR #12307 (2024-02-09): Addon Manager: Fix signals in connection check
Two changes here, to fix #11120 --
1) Refactor the signal connection and disconnection from the connection check screen to ensure a second launch of the addon manager still has the correct signals connected.
2) Switch to using a non-blocking QNetworkManager connection so that the connection check is more nearly identical to the normal way the Addon Manager accesses data from the internet.
commit 36f2f0e4 (2024-02-17): Addon Manager: Fix #11120
closed 2024-02-09 by chennes