FreeCAD/FreeCAD #11130: [Problem] Name of uninstalled addon not appear on dialog message

hasecilu, 2023-10-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When uninstalling an addon I realized that in the last successful dialog message the name of the addon was missing, instead `%s` was showed.

[image]

### Full version info

```shell
[code]
OS: Arch Linux (Hyprland/hyprland)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: makepkg
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.11.5, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.10.20a
  * Freecad-Built-in-themes-beta 1.2.2
  * WebTools 1.0.0
  * A2plus 0.4.60n
  * woodworking 0.21.30486
  * Rocket 3.2.3
  * FreeCAD_SheetMetal 0.3.1
  * kicadStepUpMod 10.20.7
  * PieMenu
  * Help 1.0.3
  * FreeCAD_FastenersWB 0.4.66
  * Assembly4 0.50.4
  * Assembly3 0.12.0
  * freecad.gears 1.0.0
  * Render 2023.8.18
[/code]
```

### Subproject(s) affected?

Addon Manager

### Anything else?

I tried this change, then run `src/Tools/updatets.py` and compile but didn't see the change (I'm missing a step or the proposed change does not affect?)
```diff
diff --git a/src/Mod/AddonManager/addonmanager_uninstaller_gui.py b/src/Mod/AddonManager/addonmanager_uninstaller_gui.py
index 86f6b9a942bc..fac0e490cdc7 100644
--- a/src/Mod/AddonManager/addonmanager_uninstaller_gui.py
+++ b/src/Mod/AddonManager/addonmanager_uninstaller_gui.py
@@ -112,7 +112,8 @@ def _succeeded(self, addon):
         QtWidgets.QMessageBox.information(
             utils.get_main_am_window(),
             translate("AddonsInstaller", "Uninstall complete"),
-            translate("AddonInstaller", "Finished removing {}").format(addon.display_name),
+            translate("AddonInstaller", "Finished removing {}").format(
+                self.addon_to_remove.display_name),
         )
         self._finalize()¿
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-02-26:
The problem here is actually a bad Spanish translation: it should not read "%s" it should read "{}", which will then get replaced as expected.

chennes, 2024-02-26:
I've fixed this on CrowdIn.

closed 2024-02-26 by chennes

hasecilu, 2024-02-27:
:scream: I was the culprit all the time...