FreeCAD/FreeCAD #11131: [Problem] Toggle visibility not working as expected

maxwxyz, 2023-10-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is a menu item / function [https://wiki.freecad.org/Std_ToggleObjects](Std_ToggleObjects) in FreeCAD which should toggle the visibility. It reverses the visibility of bodies, parts and features but that is not what the user is expecting. 

As a user, I would expect all hidden objects to become visible and all currently visible objects to be hidden.

I have attached a demo file and toggled the visibility of a body, feature, part and sketches and axis inside of them.
[visibility_toggle.zip](https://github.com/FreeCAD/FreeCAD/files/13060325/visibility_toggle.zip)

When using the Std_ToggleObjects, everything becomes hidden.
We discussed this in the forum, to implement a feature like the no show space in CATIA, to really reverse the visibility of all objects:
https://forum.freecad.org/viewtopic.php?p=714131

[TheMarkster](https://forum.freecad.org/memberlist.php?mode=viewprofile&u=19292) presented a FreeCAD macro which does work as a user would expect the Std_ToggleObjects function:
[visibility_toggler.zip](https://github.com/FreeCAD/FreeCAD/files/13060330/visibility_toggler.zip)

Can this function be implemented for the Std_ToggleObjects to work as expected from a users point of view?

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f670854
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods:

    CfdOF 1.24.5
    Curves 0.6.13
    OpenDark 2023.10.9
    sheetmetal 0.3.1
    [/code]
```

### Subproject(s) affected?

Core

### Anything else?

https://forum.freecad.org/viewtopic.php?p=714131

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2023-10-21:
Are you sure the macro does what you expect? When I load your test file, select every object of the document and then run the macro I have these issues in the tree view:
* Box item was hidden and is still hidden
* Part was visible and is still visible
* Body001 was visible and is still visible

maxwxyz, 2023-10-21:
Yes, from a user's perspective (not technically you are right):

- Box item: is a sub feature of a body, where only the last is displayed in the 3D view
- Part: needs to be visible to allow the toggle of the sub items to be visible (in contrast as it is working today)
- Body001: was visible in the tree view bot not visible in the 3D view (https://github.com/FreeCAD/FreeCAD/issues/11132)

As a user I want to see and access all items which are currently hidden. Not hide a body which was active but not visible as the last feature was visible and therefore the whole body was hidden from the user in the 3D viewport.

Then I can select the one I need, toggle with spacebar and switch back to the original visibility. I took the thing I need which was hidden (maybe a sketch or datum) and took it to the visual space in an easy click and toggle in the 3D view instead of going through each sub body in the tree view. 
That is exactly as the "show" and "no show space" in CATIA are working and a huge plus in complex models with lots of sketches, references or other parts (assemblies) http://catiadoc.free.fr/online/intug_C2/intugrf0220.htm

maxwxyz, 2023-10-21:
Maybe to clarify the issue in the example file: [visibility_toggle.zip](https://github.com/FreeCAD/FreeCAD/files/13060606/visibility_toggle.zip)

Initial situation:
[image]

When using the Macro, the visibility in the 3d view, important to the user toggles like expected:
[image]

When using Std_ToggleObjects (Toggle all objects) the result starting from the initial situation is:
[image]

What is able with the macro and should be with Std_ToggleObjects, to turn around visibility in the 3d view, not just toggle all separate features in the tree as the user is working in the 3d view.
I can switch in the no show space with the macro (make hidden objects visible in the 3d view which are currently hidden) select the sketch, press spacebar to hide it in this view and toggle the macro again.
I am there where I have started and have a visible sketch which I can use for the next operation, without even going through (in this case simple) tree view and find in which part, body and subfeature (revolve) the sketch I am looking for is hidden:
[image]

I could quickly enter the current hidden space (all invisible things in the 3d view), pick the ones I need and either copy, reference them or create binders and switch back to where I have started (in the 3d view). Another use case is to toggle the visibility look for parts, references, sketches that are currently hidden in complex parts and assemblies and toggle their individual visibility and go back to where I have started and use them as needed.
Maybe that is kind of the direction or can be implemented with @PaddleStroke in https://github.com/FreeCAD/FreeCAD/issues/10770

maxwxyz, 2025-03-09:
@pierreporte is this a relevant feature?

pierreporte, 2025-03-09:
I would say yes, but even with complex Catia model I rarely use it. It’s one of those little QOL life features that can save time in some situations.

sylanc3, 2025-05-04:
I'm a professional CATIA User and the toggle betwwen show/no show of the whole scene is a daily deal-breaker. Without that I could not work. Would be a pleasure to see the same in FC in future. Thanks for working on that!