FreeCAD/FreeCAD #11132: [Problem] Spacebar to toggle visibility of selected body/part in 3D view toggles only the last feature not the body/part

maxwxyz, 2023-10-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When selecting a part or body in the 3D view and hitting the space bar with the intent to hide it, not the body or part itself but the last feature of the selected body is hidden.
The expected behavior would be that the selected body or part is hidden.

If I want to toggle a specific feature inside a body (e.g. from the Part Design WB), I can better select this in the tree view and toggle visibility there.

When hiding parts and bodies in the tree and some parts from the 3D view it is unclear when looking at the tree view, which objects are hidden, because collapsed bodies are still shown as visible, because the space bar command only hid the last feature of the selected body in the 3D view.

In attached file I have created a body with some features and a part with a body. When selecting the first body in the 3D view (which is collapsed in the tree view) and hitting space bar or toggling the visibility, only the last feature of the body (chamfer) is hidden. The whole body is not visible in the 3D view but seems visible when looking at the tree view:
[image]

One must click through the whole tree and expand all bodies and parts to see which body is currently hidden and what feature of this body must be toggled to view it again. In this case, also the [Std_ToggleObjects](https://wiki.freecad.org/Std_ToggleObjects) does not work, separate issue for this is: https://github.com/FreeCAD/FreeCAD/issues/11131

The request for the change of the feature would be summarized:
Preferably the whole body of the selected 3D geometry in the 3D viewport is hidden, not the last feature.

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f670854
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods:

    CfdOF 1.24.5
    Curves 0.6.13
    OpenDark 2023.10.9
    sheetmetal 0.3.1
    [/code]
```

### Subproject(s) affected?

Core

### Anything else?

Separate issue as mentioned here: https://github.com/FreeCAD/FreeCAD/issues/10770
or PR https://github.com/FreeCAD/FreeCAD/pull/10805#issuecomment-1770860113

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-11-10:
Maybe this is only a problem in PartDesign WB or other situations/WB where sub features under a body are created.

adrianinsaval, 2024-02-27:
the real problem here is that when selecting something in 3d view the feature is selected and not the body itself. There are some usecases for this beahvior but it might be a good idea to make this happen only in certain situations

maxwxyz, 2024-02-27:
In part design, always the last feature is selected, which is not the expected behavior. In link branch I think the feature which created the selected geometry is selected. This makes more sense. But toggling visibility should always toggle the body.

leoheck, 2025-02-08:
Oh, this is an issue already. I thought this behavior was a pain for such a long time. Thanks @maxwxyz

leoheck, 2025-02-08:
I think the problem is the selection.
When you click on an object in the 3d view, and it belongs to a Body, It should select the Body.
This may be a particular behavior for Part Design Bodies only.

maxwxyz, 2025-02-08:
IMO not a selection problem. See https://github.com/FreeCAD/FreeCAD/issues/16164, ideally you select a fillet in the 3D view and the fillet is selected. Currently only the last feature is selected. Hide/show should only be possible on the body itself.

leoheck, 2025-02-13:
This video shows how this could work https://github.com/FreeCAD/FreeCAD/issues/19474#issuecomment-2644748437.