FreeCAD/FreeCAD #11133: [Feature Request] PD, Sketcher: Open Attachment task at Sketch creation

FlachyJoe, 2023-10-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,
At sketch creation a "Select Plane" task is opened. It's very limited as it doesn't offer to offset the plane or to rotate it and user has to change the MapMode or the Attachment after the creation.

If the Attachment task could be opened it would improve the workflow.

### Full version info

```shell
[code]
OS: Debian GNU/Linux trixie/sid (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34594 (Git)
Build type: Unknown
Branch: main
Hash: 3d9e3fa809e7dbc3ec60ed23195759b5aff4ed54
Python 3.11.6, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-10-27:
This is the current dialog in RT link daily. When creating a sketch and selecting a reference, the task view show the sketch attachment dialog:
[image]

maxwxyz, 2023-12-11:
@PaddleStroke is that something you could implement, as you are currently fixing ux things in sketcher and generally adding improvements to sketcher?

pierreporte, 2023-12-15:
Just tested this using the attachment panel with a sketch, via the property panel (Map Mode). It is more powerful than I thought when discussing about #11604 because it allow to set the center of the sketch. However, it does not allow two things that would be very useful, not only to sketches but everything that uses the attachment panel:

1. It is impossible to set the sketch origin it using a base plane and a line (datum or from a sketch) intersecting the plane (the sketch center being the intersection).
2. It is impossible to set the sketch axis to follow existing geometry. It could allow to chose along which line gives the direction of the X axis. For the Y axis it could be the same but a flip button would be sufficient (mathematically there is only two possibilities when the support plane and the first axis are set).

maxwxyz, 2024-03-13:
Added info from the closed duplicates. This would all be resolved if the attachment dialog is opened at new sketch creation.

**Can’t create a sketch on a face when PartDesign_NewSketch is active**
If you select a face, then click new sketch, it works.
If you select new sketch, then select a face, you cannot validate.
To create a sketch on the face of a part, it is necessary to:
- click the face,
- click on PartDesign_NewSketch.

The tool should allow the opposite as well:
- click on PartDesign_NewSketch,
- click on the face.
This is how the dialog currently works and you cannot select different part of an object but only planes:
[image]

maxwxyz, 2024-04-13:
To add from the FreeCAD Day 2024 Complaint Session:
**Selecting coordinate system for a sketch is convoluted, there are multiple ways to do it, when editing a sketch the coordinate system is not visible.**
This issue is linked to the first mention.
Discussion: 
It is absolutely absurd that the sketcher has two different methods to attach a sketch. When you start, depending on if you are in a sketcher or Part Design and they're both bad. When you start a sketch, you're supposed to choose a plane. Or if you attach it to a face they're different to each other and they're both bad.
Selecting the coordinate system for the sketcher is confusing, not obvious. There's two ways to do it, and they're incompatible, and neither of them is what you actually want.

Related: 
- https://github.com/FreeCAD/FreeCAD/issues/12106
- https://github.com/FreeCAD/FreeCAD/issues/12584

pierreporte, 2024-05-07:
I proposed an alternative way in #12106, that is clearer for how sketches are positioned.

PaddleStroke, 2024-11-22:
ok I'm on it. Shouldn't be too hard.

maxwxyz, 2024-11-22:
@PaddleStroke great, maybe with #12106 or is this too unrelated?

FlachyJoe, 2025-04-18:
Closed by #18441

maxwxyz, 2025-08-13:
@PaddleStroke would you mind creating a PR enabling this by default (and if possible, also show if the user has a preselection)

PaddleStroke, 2025-08-20:
Changing the default is trivial. But I fear user backlash. This needs detailed discussion by the DWG and perhaps even forum?

PaddleStroke, 2025-08-20:
Preselection is a first thing to fix obviously.

maxwxyz, 2025-08-20:
The current way is basically hidden from the user - manually create a parameter, but the issue is closed. Maybe we can create a preference, set its default to use the attachment dialog and then users can easily change the behavior.
@FreeCAD/design-working-group any opinions?

pierreporte, 2025-08-20:
The best would be to do a new attachment panel for sketches as I proposed, and apply it to `Sketcher_NewSketch` so that the PD command is not needed anymore. See https://github.com/FreeCAD/FreeCAD/issues/12106 for both the new panel and the deduplication. It would allow current workflow and better attachment. Technically every time the attachment panel is used it should be tailored to the thing being worked on but it’s not mandatory to do all of them at once.

But if we’re in a hurry for 1.1, then a user preference will be welcome.

PaddleStroke, 2025-08-21:
I agree that I find the current attachment task not good enough to be set as default as of now. My plan was to change the default after reworking the task panel. Which I am not sure when I can tackle.

maxwxyz, 2025-09-10:
I'll reopen then so this won't get lost.

BTW I noticed an issue with either this dialog or the datum sizes:
1. when the flag is set to open the attachment dialog on sketch creation, the datums are the size of the entire 3D view
2. when editing the attachment of the same sketch later, they are the correct size

Demo:

https://github.com/user-attachments/assets/8405d45a-e94a-4ee1-949a-8cd33dfa5df7

@PaddleStroke @kadet1090

Krrish777, 2026-01-03:
Has this issue been resolved?