FreeCAD/FreeCAD #11135: [Feature Request] PD: Pad UpTo multiple faces

FlachyJoe, 2023-10-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

PartDesign_Pad only allow to select one face to end to. In cases of the base's projection is partially out of one face it's not sufficient (see #11111 and #10538).

### Full version info

```shell
[code]
OS: Debian GNU/Linux trixie/sid (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34594 (Git)
Build type: Unknown
Branch: main
Hash: 3d9e3fa809e7dbc3ec60ed23195759b5aff4ed54
Python 3.11.6, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

OCC take a shape as limit:
`BRepFeat_MakePrism::Perform (const TopoDS_Shape& Until)`
but we provide a face:
```
void FeatureExtrude::generatePrism(TopoDS_Shape& prism,
                                   const std::string& method,
                                   const TopoDS_Shape& baseshape,
                                   const TopoDS_Shape& profileshape,
                                   const TopoDS_Face& supportface,
                                   const TopoDS_Face& uptoface,
                                   const gp_Dir& direction,
                                   PrismMode Mode,
                                   Standard_Boolean Modify)
// […]
        BRepFeat_MakePrism PrismMaker;
// […]
            PrismMaker.Perform(uptoface);
```

We should create a TopoDS_Shape from selected faces and provide it to generatePrism.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-11-13:
what's the next step @FlachyJoe ? from a UX perspective I think it would be ideal to unify up to face and up to shape into just one mode where we can select either an entire shape or a group of faces. I can help out with the UI when we're settled on the App side

maxwxyz, 2023-11-13:
is up to shape now also integrated into up to last and up to first, when the profile is intersecting multiple faces (https://github.com/FreeCAD/FreeCAD/issues/11111#issuecomment-1772432318)?

FlachyJoe, 2023-11-13:
@maxwxyz No, uptolast and uptofirst still use faces and IMHO this behavior shouldn't changed. With these two modes the prism ends at the face **continuity** and this feature is required. The uptoshape mode use all the faces of the base shape and doesn't extend them.
@adrianinsaval 
UpToShape could become the default mode for sketch which doesn't belong a face of the base shape.
UpToShape could allow another shape selection, not only the base shape. UptoFace could use multiple faces. But I think we can't merge them because UpToShape can't deal with offset.

adrianinsaval, 2023-11-13:
If we get multiple faces working, will there be any remaining usecase for up to shape? I don't think so.

FlachyJoe, 2023-11-14:
@adrianinsaval In fact I don't see a case where multiple faces are needed but not the full shape.

adrianinsaval, 2023-11-14:
> @adrianinsaval In fact I don't see a case where multiple faces are needed but not the full shape.

if you need to select any intersected set of faces besides the first ones, a pocket up to shape would be pretty pointless too

maxwxyz, 2023-11-14:
> @adrianinsaval In fact I don't see a case where multiple faces are needed but not the full shape.

what about the following example file: 
[pad.zip](https://github.com/FreeCAD/FreeCAD/files/13347904/pad.zip)

If I extrude a profile which would intersect multiple faces. What would happen when I select up to shape? I need multiple faces to constrain it to either the first shell walls (inner or outside) or the last shell wall (inner or outside) or the wall in the middle. To first and to last currently does not work at all in this situation.
[image]

FlachyJoe, 2023-11-14:
@maxwxyz I agree it's a use case. Solved by #11392 
[image]
[image]
[image]
One face selection conserves the previous behavior:
[image]

maxwxyz, 2023-11-14:
@FlachyJoe and what is the result with up to shape? through all or to the first? In this example to last and to first does not work.

FlachyJoe, 2023-11-15:
@maxwxyz When you select a full shape it ends at the first intersection for Pad and at the last for Pocket (see https://github.com/FreeCAD/FreeCAD/pull/11392#issuecomment-1809974237).

I think I can make a switch for this and base UpToFirst and UpToLast to shape and not face. But it will break the backward compatibility as the face extension won't occurs. Do we need an UpToLastFace and UpToFirstFace? I open [a forum thread](https://forum.freecad.org/viewtopic.php?t=82724).

FEA-eng, 2024-06-13:
Perhaps this issue could be closed as Up to Shape is fully merged now and allows face selection in GUI.

FlachyJoe, 2024-06-13:
Buggy (#14663) but done :smiling_face_with_tear:

closed 2024-06-13 by FlachyJoe