FreeCAD/FreeCAD #11142: [Problem] Multiscreen under windows 11 : menus open in wrong screen

OlivierHersent, 2023-10-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Reproduction : set-up a multi-screen configuration with FreeCad running on top screen. FreeCAD menus e.g. "Help" open on the bottom screen.

### Full version info

```shell
0.21.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-22:
Full version info please. See [About](https://wiki.freecad.org/About).

luzpaz, 2023-10-24:
@OlivierHersent please follow-up

OlivierHersent, 2023-10-24:
Version 0.21.1

OlivierHersent, 2023-10-24:
Sorry it took me a while to realize that there is a "copy to clipboard" button.
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

OlivierHersent, 2023-10-24:
When you test, also with the Windows screen relocate shortcut (WIN + Shift up + >) you will see multiple issues related to menu location, font size, and icon sizing. Probably the screen repaint primitives do not take the right screen as reference for location and resolution.

luzpaz, 2023-10-24:
Can you test on the dev version and report back ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

luzpaz, 2023-10-27:
@OlivierHersent ping

maxwxyz, 2023-10-28:
Can confirm. 0.22dev 2023-10-26 2weekly.
Preferences window will open on the display where FreeCAD opened the first time on a setup with three monitors.

OlivierHersent, 2023-10-28:
Testing on latest build, much better behavior. I could reproduce the issue only by manually dragging the main window between two screens with different resolutions (small laptop & large second screen). At some point the window on large main screen switches to the resolution of the smaller screen, so everything appears huge. At this point the menus pop-up pretty much randomly, undocked.
but again switching screens with WIN/SHIFT/RIGHT works fine, so this is only a edge case now in my experience with latest build.

[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34789 (Git)
Build type: Release
Branch: main
Hash: 41503d63ebc3de96d156e6eb0cdb889ebb150bd4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.13
[/code]

maxwxyz, 2023-10-30:
Not an edge case! At work I have three displays, at home 2. 
Opening Tools>Edit Parameters at home renders FreeCAD unusable as the window is opened on the third "missing" monitor. No chance to get it with Win + Shift + Right / Left / Up to the right window.
I have to close FreeCAD (with task manager) and cannot save --> loosing all progress.
At home I cannot work with FreeCAD on the same machine, it somehow saved the third monitor as its place to open new windows.

Confirmed with current weekly version:
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34810 (Git)
Build type: Release
Branch: main
Hash: 196cbecde66a1c3a1b56b8c532ad1faedff4818b
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.7
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * Defeaturing 1.2.1
  * fasteners 0.4.67
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.22
  * sheetmetal 0.3.1
  * Silk 0.1.5
[/code]

maxwxyz, 2024-01-12:
This is really annoying, I have no third monitor at home so I cannot access things like the "Edit parameters..." dialog at all :(

xtemp09, 2024-01-14:
@maxwxyz, you should create an issue so your complaint won't be lost.

maxwxyz, 2024-01-14:
@xtemp09 isn't this an issue about the same topic? I was just adding more examples.

xtemp09, 2024-01-14:
Menus activated at the bottom screen instead of the top one and a window shown off-screen. From my point of view they are not connected. =)

But of course, it's for you to decide.

luzpaz, 2024-08-30:
Can someone test on Qt6 ?

luzpaz, 2024-10-27:
@maxwxyz any chance you can test this on Qt6 ?

maxwxyz, 2024-10-27:
Unfortunately no.

luzpaz, 2025-01-13:
I think we may need a multi-monitor tag

closed 2025-06-30 by chennes