FreeCAD/FreeCAD #11151: [Problem] If several instances of PropertyFileIncluded reference the same file then this could lead to loss of data

wwmayer, 2023-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://forum.freecad.org/viewtopic.php?p=714526#p714526

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34729 (Git)
Build type: debug
Branch: issue_11113
Hash: 0b0e9effeb42c9ad441e1a70efcbfb2cd3d05223
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jesuscastane, 2026-09-17:
it seems it remains a bug on 1.1.3

[image]

ChickenBaconOnion, 2026-10-10:
Remains in 1.1.4 as well. Extremely annoying, happens often. Thankfully AI provided me with a way to restore the svg templates, otherwise a lot of work would be lost.

[image]