FreeCAD/FreeCAD #11154: [Problem] Material: Render workbench unable to work with current models in dev

davesrocketshop, 2023-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Render workbench is experiencing issues interacting with the new material subsystem.

See:
https://github.com/FreeCAD/FreeCAD-render/issues/312
https://github.com/FreeCAD/FreeCAD-render/pull/318

### Full version info

```shell
Version 0.22 (dev)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2023-10-24:
The legacy material system was simple with few rules. Render ignored them all.

Material cards simply do not follow the documented standards. Some are understandable, such as extra rendering parameters and it should be easy to expand the parser to support them.

Others are head scratching, such as color attributes in the architectural section instead of the rendering section.

Maintaining compatibility with this old behaviour should be possible, but I don't really understand why this situation exists. Bringing it in line with the new material system is going to require far more effort than it should.

Getting the current software to work with the new system is the priority. Fixing the workbench will follow later.

luzpaz, 2024-08-25:
CC @howetuft if you don't mind, i'm adding you to this ticket so you can get updates as well.