FreeCAD/FreeCAD #11163: [Problem] The Path direction marker is too small on high resolution displays

sliptonic, 2023-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hovering over a toolpath marker shows a small arrow indicating the direction of travel for the tool.  On high resolution monitors, the arrow is too small to see and hidden behind the mouse pointer.

The arrow size is not a configurable options

### Full version info

```shell
[code]
OS: Linux Mint 20 (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34757 +1 (Git)
Build type: Unknown
Branch: (HEAD detached from upstream/pr/11108)
Hash: 2118faddba1f07b8eb2598a5db9d8f6360cbace7
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * OpenDark 2023.10.9
  * fasteners 0.4.67
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
  * Ondsel-Lens 2023.8.23.02
  * FeedsAndSpeeds 0.5.0
  * Assembly3 0.12.0
[/code]
```

### Subproject(s) affected?

Path

### Anything else?

This is more than cosmetic since the indicator is used to validate tool direction and to debug path creation.  

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-27:
Any proposals for how to improve ?

luzpaz, 2023-11-17:
@MisterMakerNL perhaps you can weigh-in ?