FreeCAD/FreeCAD #11164: [Feature Request] Sketcher: Dimensioning Angles by Three Points

maxwxyz, 2023-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The contextual dimension tool as well as the angle tool should allow the constraint/dimensioning of angles by selecting three points.
_Comment by @PaddleStroke in https://github.com/FreeCAD/FreeCAD/issues/10605#issuecomment-1776871732_

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-10-26:
This one is more difficult because currently the angle constraint does not support 3 points. So this needs to be implemented upstream before Sketcher_Dimension can do this.
Well the tool could do it by creating construction lines I guess but that's a bit dirty imo.

maxwxyz, 2023-10-27:
> Any relationship to #7221 ?

no

> This one is more difficult because currently the angle constraint does not support 3 points. So this needs to be implemented upstream before Sketcher_Dimension can do this. Well the tool could do it by creating construction lines I guess but that's a bit dirty imo.

I cannot code but can I help somehow?

PaddleStroke, 2023-10-27:
> Any relationship to https://github.com/FreeCAD/FreeCAD/issues/7221 ?

No but https://github.com/FreeCAD/FreeCAD/pull/11174 fixes #7221

> I cannot code but can I help somehow?

I'm not sure how. I don't know well how to create new constraints. This is something that @FlachyJoe has done in the past (circle to circle distance). Perhaps he'll be interested to add this to his targets?

FlachyJoe, 2023-10-28:
I add this to my Todo List