FreeCAD/FreeCAD #11166: [Feature Request] Add Preferences for Multiple Lights in the Scene

maxwxyz, 2023-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

To extend the functionality of https://github.com/FreeCAD/FreeCAD/issues/11113 implemented by https://github.com/FreeCAD/FreeCAD/pull/11146 it would be great to add, adjust, and remove lights in the preference dialog of FreeCAD.
The arrows could indicate a light each.
[image]

The direction of the lights should be editable as well as the intensity. If possible, even the color could be adjusted and all saved in the preferences. The current light which is edited could be highlighted in the view, maybe even represent color and intensity on the ball (like a real time preview). It should be possible to add new lights as well as to remove them again.
Maybe themes could provide settings of the default lightning.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-10-24:
I would add that the easy switch between light settings, like MisterMakerNL showed in [the screenshot](https://github.com/FreeCAD/FreeCAD/pull/11146#issuecomment-1775520510), would be of great value.

maxwxyz, 2023-10-24:
@MisterMakerNL could you specify what settings or presets this would include?

obelisk79, 2023-10-25:
No, I don't necessarily see how this would directly impact my theme. I do worry about over-complicating with too many advanced controls.

What is the problem adding multiple lights like this is intended to solve?

maxwxyz, 2023-10-26:
The background in the preferences is currently all white. Maybe a theme can set it to the 3D view background. Or is this feature able to pull the 3D view background @wwmayer ?

luzpaz, 2023-11-20:
Bumping to continue the conversation

luzpaz, 2023-11-26:
@maxwxyz can you explicitly elaborate on why this multiple light feature is useful ? TIA

maxwxyz, 2023-11-27:
@luzpaz as @xtemp09 mentioned above https://github.com/FreeCAD/FreeCAD/pull/11146#issuecomment-1775520510
It is useful to see the object. One light a shiny objects can make the parallel faces facing the camera plane all seem white (see #11113 :
[image]
viewing from top:
[image]

When sketching or selecting references you do not see differences in face color or selecting color so it gets pretty hard with assemblies or referencing in sketches. The fix #11146 lets turn the default position of the light which fixes this issue, but with only one light present in the scene and no global/background illumination, all faces not directly illuminated by the one light appear black.
Simple model to show the problems of the fix of the first problem:
[image]

Constantly adjusting the default main headlight position is not ideal in a workflow. Other CAD applications have a general background illumination or multiple light, where you can even switch to different presets or even an HDRI as background image (for reflections).
If multiple lights is too difficult to add, maybe a slider for a general background light from all directions would be great. It is not great if the one setup light creates white faces and completely black faces in the model which makes it hard to work with.

The presets @MisterMakerNL referenced are also a great addition and maybe even themes can ship their individual light settings, to display the model in a great light and much nicer user interface

[image]

Example of multiple light settings in CATIA: http://catiadoc.free.fr/online/basug_C2/basugbt1701.htm

maxwxyz, 2023-12-08:
In addition, I propose to change the color / material appearance of the current sphere to be able to see the light source and shadow area better.
current look (sphere is small in the dialog with a huge arrow and not much difference between light and shadow on the sphere):
[image]

proposed preview (screenshot as implemented in the material editor):
[image]

```
Values:
Ambient color: #333333
Diffuse color: #cccccc
Emissive color: #000000
Specular color: #cccccc
Shininess: 0.6
Transparency: 0
```

maxwxyz, 2024-01-08:
Additionally, the ambient color and ambient light intensity should be a preference and an ambient light should be in the 3D scene by default so completely dark parts (shadow of headlight) are not created and everything should be visible. the ambient color should be 0.2 0.2 0.2 as a default.

maxwxyz, 2024-02-13:
I've noticed, that the 'Reset' button for this preferences panel also does no work.

M4xTheMan, 2024-10-08:
Could this be closed too or is it still an issue?

maxwxyz, 2024-10-09:
IMO yes, there is no possibility to add multiple lights, change individual lights, use background illumination or even an HDRI

M4xTheMan, 2024-10-09:
Alright, thank you!

maxwxyz, 2025-01-22:
Demo of how this could look with multiple lights added via addon (should be possible with Python I guess?)
Would also fix: https://github.com/FreeCAD/FreeCAD/issues/14965

https://www.youtube.com/watch?v=E75hptWOSZA

[image]

@obelisk79 @kadet1090 FYI, how complex would such an implementation be?

kadet1090, 2025-01-22:
Given that this can be done in addon - should be pretty easy to do as we already have everything needed in code. Hardest to do would be UI for this.

maxwxyz, 2025-01-22:
I mean ideally we would just have a three point light setup as default without a UI in the beginning. Then to adjust it use the existing preference page for light sources?

luzpaz, 2025-01-22:
cc @oficinerobotica just wanted to put this on your radar (since your tutorial a model are being referenced here)

M4xTheMan, 2025-02-22:
@kadet1090 ping :-)

closed 2025-02-24 by chennes