FreeCAD/FreeCAD #11168: [Problem] Addon Manager: local cache refresh issue, and other

kaktusus, 2023-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Refresh of local cache not proceeding correctly and cannot finish ...

[image]

This issue he has been observing for several releases.

```
17:09:08  Loading Assembly workbench...
17:09:14  flag
17:09:14  flag2
18:45:31  Previous cache process was interrupted, restarting...
18:45:35  libpng warning: iCCP: known incorrect sRGB profile
18:45:35  libpng warning: iCCP: known incorrect sRGB profile
18:46:20  Timeout while fetching metadata for macro Zoom1 1
18:46:25  libpng warning: iCCP: known incorrect sRGB profile
18:46:28  libpng warning: iCCP: known incorrect sRGB profile
18:47:23  libpng warning: iCCP: known incorrect sRGB profile
18:47:32  Timeout while fetching metadata for macro Honeycomb
18:47:39  Timeout while fetching metadata for macro Make Cube
18:47:45  Timeout while fetching metadata for macro Site From Contours
18:48:05  Timeout while fetching metadata for macro Bevel
18:49:23  libpng warning: iCCP: known incorrect sRGB profile
19:47:41  Timeout while fetching metadata for macro Align Face Object to View
19:47:41  Unable to open macro wiki page at https://wiki.freecad.org/Macro_Align_Face_Object_to_View
19:47:41  Timeout while fetching metadata for macro Align View to Face
19:47:41  Unable to open macro wiki page at https://wiki.freecad.org/Macro_Align_View_to_Face
19:47:41  Timeout while fetching metadata for macro Copy3DViewToClipboard
19:47:41  Unable to open macro wiki page at https://wiki.freecad.org/Macro_Copy3DViewToClipboard
19:47:41  Timeout while fetching metadata for macro FCCamera
19:47:41  Unable to open macro wiki page at https://wiki.freecad.org/Macro_FCCamera
19:47:41  Timeout while fetching metadata for macro Mouse Cross
19:47:41  Unable to open macro wiki page at https://wiki.freecad.org/Macro_Mouse_Cross
19:47:41  Timeout while fetching metadata for macro Rotate View
19:47:41  Unable to open macro wiki page at https://wiki.freecad.org/Macro_Rotate_View
20:09:14  Previous cache process was interrupted, restarting...
20:09:23  Error connecting to the Wiki, FreeCAD cannot retrieve the Wiki macro list at this time
```

In the log, among other things, we see references to the old domain.

---

Yet another dysfunction.
Python dependencies

No update capability. 
The program is terminated unexpectedly.

[vokoscreenNG-2023-10-24_20-17-10.webm](https://github.com/FreeCAD/FreeCAD/assets/39754351/ba0165ee-17fc-4e7b-9c62-184d384c2fa4)

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34729 (Git) AppImage
Build type: Release
Branch: main
Hash: 0b0e9effeb42c9ad441e1a70efcbfb2cd3d05223
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * Curves 0.6.13
  * sheetmetal 0.3.1
  * Rocket 3.2.3
  * kicadStepUpMod 10.20.7
  * fasteners 0.4.67
  * Render 2023.8.18
  * BIM 2021.12.0
  * FreeCAD-Reinforcement-added_translations.zip
  * FreeCAD-Reinforcement-master.zip
  * Reinforcement
  * Alternate_OpenSCAD 1.0.0
  * Woodworking 0.21.30486
  * dodo 1.0.0
  * Alternate_OpenSCAD.backup1691601718.9594336.zip
  * OpenDark 2023.10.22
```
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kinghat, 2023-11-06:
i dont think its just when refreshing the local cache but also when it just checks for updates. the majority of the time it just hits a spot where it hangs and never finishes:

[image]

[image]

> OS: KDE Flatpak runtime (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.21.1.33694 (Git)
> Build type: Release
> Branch: (HEAD detached at f670854)
> Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
> Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
> Locale: English/United States (en_US)
> Installed mods: 
>   * BIM 2021.12.0
>   * ThreadProfile 1.89.0
>   * fasteners 0.4.70
>   * ExtremeProDark 2.9.0
>   * Manipulator 1.5.0
>   * OpenDark 2023.10.30
>

chennes, 2024-01-30:
I think I've tracked down this cache refresh issue, it should be addressed by #12194.

kaktusus, 2024-02-02:
```
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35759 (Git) AppImage
Build type: Release
Branch: main
Hash: 90d36f60c30cb6549a7987fcd9d4769eab1c160c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * Curves 0.6.23
  * sheetmetal 0.4.0
  * Rocket 3.3.0
  * kicadStepUpMod 10.23.0
  * fasteners 0.5.0
  * Render 2024.1.23
  * BIM 2021.12.0
  * FreeCAD-Reinforcement-added_translations.zip
  * FreeCAD-Reinforcement-master.zip
  * Reinforcement
  * Alternate_OpenSCAD 1.0.0
  * Woodworking 0.21.30486
  * dodo 1.0.0
  * Alternate_OpenSCAD.backup1691601718.9594336.zip
  * OpenDark 2023.12.17
  * freecad.gears 1.0.0
  * Movie 2023.6.10
```
Local cache refresh is working ...

closed 2024-02-02 by kaktusus