FreeCAD/FreeCAD #11169: [Feature Request] Adjust Clipping Plane Position/Direction by Selection of References

maxwxyz, 2023-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently it is only possible to set three individual clipping planes normal to the base axes (x, y, z).
It is also possible to clip in a custom direction with an offset, by specifying a vector.

It would be beneficial to create the clipping plane (or fill in the direction vector and offset) by clicking on a reference in the 3d view
- a face/plane or datum for their normal direction and offset
- a line or axis for their direction and offset 0 or at the location the mouse pointer is clicking
- a vertex or point to get the offset 

It would be great if the offset and maybe direction or rotation could be edited after the selection of the reference (to go through a model or assembly by sliding around the specific clipping plane).

It would be great if persistent clipping planes could be saved in the model tree and be activated (e.g. by a double click and loaded into the clipping plane tool). Maybe even attach them to references permanently or move them interactively in the 3d view like in other CAD software. --> all future, separate FR :)

I know that there is a persistent section cut but this adds complexity by real geometry in the tree and the Part_SectionCut is a heavy operation and takes long to load on complex geometry or assemblies.

[image]

And there is also the long wished closing of the hollow shells of the clipping planes https://github.com/FreeCAD/FreeCAD/issues/5570

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Anything else?

I have found more about clipping planes in OCCT:

https://dev.opencascade.org/doc/overview/html/occt_user_guides__visualization.html#occt_visu_4_4_16
https://dev.opencascade.org/doc/refman/html/class_graphic3d___clip_plane.html

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-10-24:
I think it should be possible to get the normal vector from a selected face directly from coin so implementing this might be possible without introducing dependency to the Part module.

the occt clipping plane stuff is probably not usable since we use coin3d not occt's rendering

maxwxyz, 2023-10-25:
I've created a separate FR for the interactive sliding of clipping planes as this is currently already implemented in the FEM WB: https://github.com/FreeCAD/FreeCAD/issues/11173

Mannshoch, 2024-08-05:
Related in FC Forum: [Request for Merging feature: "Persistent Section Cut" and "Clipping Plane" and FEM clippling Plane](https://forum.freecad.org/viewtopic.php?p=772286#p771295)

maxwxyz, 2024-09-01:
Made a separate issue for saving/storing custom and multiple clipping planes / section views:
- https://github.com/FreeCAD/FreeCAD/issues/16187

maxwxyz, 2024-09-01:
More or less it could be also defined like the Attachment dialog, as it is just a plane after all, so also by one axis and a point, three points,... just the direction is needed.

CATIA Example:
[image]

Mannshoch, 2024-09-02:
How solid works do It:
Solidworks Section view:  https://youtu.be/P6C3MZW9FLM
Solidworks Help: https://help.solidworks.com/2024/english/SolidWorks/sldworks/t_section_views_models.htm