FreeCAD/FreeCAD #11170: [Problem] GUI: New (freed up) workbench selector.

kaktusus, 2023-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have been observing this problem since the workbench selector was rebuilt.
I am happy to be able to place it in the desired location. 
On my Debian I have several virtual desktops.

This widget seems to live its life. Check out the video.

[vokoscreenNG-2023-10-24_20-45-33.webm](https://github.com/FreeCAD/FreeCAD/assets/39754351/4e70f11e-0046-483d-986e-309628d721f5)

All I have to do is move it a little and everything starts working fine. However, after some time it starts to be freed up again.
Other strange behavior is that it self-closes after it is expanded. You can't see it now, however, it happens quite often.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34729 (Git) AppImage
Build type: Release
Branch: main
Hash: 0b0e9effeb42c9ad441e1a70efcbfb2cd3d05223
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * Curves 0.6.13
  * sheetmetal 0.3.1
  * Rocket 3.2.3
  * kicadStepUpMod 10.20.7
  * fasteners 0.4.67
  * Render 2023.8.18
  * BIM 2021.12.0
  * FreeCAD-Reinforcement-added_translations.zip
  * FreeCAD-Reinforcement-master.zip
  * Reinforcement
  * Alternate_OpenSCAD 1.0.0
  * Woodworking 0.21.30486
  * dodo 1.0.0
  * Alternate_OpenSCAD.backup1691601718.9594336.zip
  * OpenDark 2023.10.22
```
```

### Subproject(s) affected?

None

### Anything else?

https://discord.com/channels/870877411049357352/1155190517223657702/1160512535678693376

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-29:
@PaddleStroke workbench selector regression here

PaddleStroke, 2023-10-30:
Hmm I am not sure to understand precisely the situation.
It seems to me that you are putting your workbench selector as a free-floating widget. Which means that if you move your freecad window the widget does not move along. It is not docked in a dock area.

The bad behavior is that it is not disappearing when you change of software right? I don't see where this could come from. Do you know precisely when it started to behave like this?

kaktusus, 2023-10-30:
Exactly the point is that if FreeCAD is not active this widget should be hidden.
such behavior occurs from the very beginning when the new functionality was introduced.

PaddleStroke, 2023-10-30:
What new functionality are you refering to ?

kaktusus, 2023-10-30:
just about the fact that I can put the workbench selector anywhere :wink:

PaddleStroke, 2023-10-30:
Ah, but isn't that around for a very long time? I have no idea when this got introduced.

luzpaz, 2023-10-30:
> I have been observing this problem since the workbench selector was rebuilt.

@PaddleStroke wasn't this done recently ?

kaktusus, 2023-10-30:
This is a relatively new functionality ...
Of course, the passage of time is a relative concept and everyone perceives it individually. :laughing:

PaddleStroke, 2023-10-30:
I must be wrong then. I have no clue about the topic then sorry

luzpaz, 2023-10-31:
CC @wwmayer,, not sure but this issue may be related to one of our [PRs](https://github.com/FreeCAD/FreeCAD/issues/11170#issuecomment-1785981367)

wwmayer, 2023-10-31:
Some time ago there was a change to put the workbench selector to the left or right side or put it to a tool bar. But don't know if that's related to this issue.

> not sure but this issue may be related to one of your https://github.com/FreeCAD/FreeCAD/issues/11170#issuecomment-1785981367

Your link references this bug report. But it's unlikely that it's one of my PRs because I still prefer and use the classical workbench selector, i.e. the combo box in the middle of the tool bar and I can't remember that I have changed that.

I guess you meant https://github.com/FreeCAD/FreeCAD/pull/7679

PaddleStroke, 2023-10-31:
> I guess you meant https://github.com/FreeCAD/FreeCAD/pull/7679

That's what I thought but no the issue is not about this cornerWidget PR. The user is not using it, he's putting the combobox as a floating widget. ie not attached to any dock area.

adrianinsaval, 2023-11-14:
can you repropoduce on a clean config?

kaktusus, 2023-11-14:
I'll check it for you in a few hours when I finish work.

kaktusus, 2023-11-17:
I'm sorry you had to wait so long. :disappointed: 

In the video, among other things, I reset all FreeCAD settings from the preferences panel.
 `-` Without the expected effect.
I removed FreeCAD configuration files from my system.
 `-` It worked until I restarted the program.

[vokoscreenNG-2023-11-17_17-04-04.webm](https://github.com/FreeCAD/FreeCAD/assets/39754351/1a5079fc-4460-4ac9-8379-ef63633fd677)

[vokoscreenNG-2023-11-17_17-20-18.webm](https://github.com/FreeCAD/FreeCAD/assets/39754351/eaffc49a-7a39-4b57-9942-ec97b2d9094e)

kaktusus, 2023-11-20:
Another nasty anomaly is occurring.
However, it is relatively rare and difficult to repeat.

[vokoscreenNG-2023-11-19_17-14-20.webm](https://github.com/FreeCAD/FreeCAD/assets/39754351/75da8386-a8c9-45ae-97f6-ad7df874d495)

In the video, you can see how the drop-down list opens when clicked and immediately closes on its own.
This problem I noticed a long time ago, however, now I managed to record it.

luzpaz, 2024-08-25:
Is this still relevant ?

kaktusus, 2024-08-25:
Hi L.

No. The functionality has been changed and it is no longer possible to pull this object outside the FreeCAD window area.

closed 2024-08-25 by kaktusus