FreeCAD/FreeCAD #11173: [Feature Request] Interactive Clipping Plane (like implemented in FEM WB)

maxwxyz, 2023-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The current clipping plane options in the core module is quite static.
In the FEM WB there is a feature to create a new clipping plane on a face (selection). Can these features be implemented in the core clipping plane dialog?
[image]

In FEM WB, multiple clipping planes can be activated on different face selections.
All clipping planes can be manipulated directly in the 3D view to slice/go through the whole model. (I think with Coin3D's SoClipPlaneManip: https://www.coin3d.org/Coin/html/classSoClipPlaneManip.html)
[image]

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Anything else?

https://www.coin3d.org/Coin/html/classSoClipPlaneManip.html
https://www.coin3d.org/Coin/html/classSoClipPlaneManip.html#a1fdc2a28dece3ca5a1fc5edcc6972713

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2023-10-25:
Can we simply replace current clipping feature with that from FEM? It seems to be more polished and there is no reason to keep both of them in the program.

maxwxyz, 2023-10-25:
In the current tool you can specify an individual vector and offset. In FEM you need a face for a reference. Maybe the FEM clipping plane can intelligently merged with the current design, to replace the base axes which make the dialog so big.

luzpaz, 2023-10-27:
needs more core dev feedback here

maxwxyz, 2024-03-09:
Related: https://github.com/FreeCAD/FreeCAD/issues/6204

Mannshoch, 2024-08-05:
Related in FC Forum: [Request for Merging feature: "Persistent Section Cut" and "Clipping Plane" and FEM clippling Plane](https://forum.freecad.org/viewtopic.php?p=772286#p771295)

maxwxyz, 2025-07-20:
In RT branch you can adjust (translate and rotate) the clipping plane via dragger (and even save the views):

[image]