FreeCAD/FreeCAD #11180: [Path] Boundary Dressup fails on Array

morganrallen, 2023-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Unlike many other Ops, the Array uses a (Python) List/Array as it's Base Property. This Causes the Boundary Dressup to fail when trying to determine the Safe Height.

```
  File "/usr/lib/freecad/Mod/Path/Path/Dressup/Boundary.py", line 159, in execute
    self.safeHeight = float(PathUtil.opProperty(self.baseOp, "SafeHeight"))
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'TypeError'>: float() argument must be a string or a real number, not 'NoneType'
```

I'm not sure what the best solution is, my initial thought would be if it's a List, iterate through and use the largest SafeHeight value. If this sounds reasonable I can put together a patch.

### Full version info

```shell
[code]
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34762 (Git)
Build type: Unknown
Branch: main
Hash: b70956ba3dd0ffb12bac6761d49dcacb5de8c950
Python 3.11.5, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * Render 2023.8.18
  * Plywood_WB
  * Launcher
  * BIM 2021.12.0
  * Curves 0.6.13
  * sheetmetal 0.3.1
  * A2plus 0.4.60n
  * Trails 2022.1.0
  * kicadStepUpMod.backup1663197104.2499094
  * SelectorToolbar
  * Behave-Dark-Colors 0.1.1
  * kicadStepUpMod 10.20.2
  * Assembly4 0.50.4
[/code]
```

### Subproject(s) affected?

Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2023-12-05:
@morganrallen My one person opinion is that is the approach that makes sense.  If you have an array of parts you are setting up a path for, the safe height has to be above all of them or it isn't safe.  So the logic resonates for me.

musthavero, 2024-03-22:
Just bumped into this issue, so it's still there. But I would change the Array, not the Boundary. The Array has to have just one maximum safe height and clearance height. otherwise other operations applied to the Array will also fail.

luzpaz, 2025-01-28:
@opensourcecnc make a PR ?

closed 2025-06-23 by github-actions[bot]