FreeCAD/FreeCAD #11183: [Notification Area] Repeated messages

xtemp09, 2023-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I launched _FreeCAD_, loaded a file and as you can see in the screenshot, there are many repeated messages in the _Notification Area_.

<details><summary>Screenshot 1</summary>

[image]

</details>

My proposal: _FreeCAD_ shows one message and say that the last message has been repeated 10 times.

----

I set the message limit to **5**.

<details><summary>Screenshot 2</summary>

[image]

</details>

It did not influence the number of messages. After restart:

<details><summary>Screenshot 3</summary>

[image]

</details>

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34784 (Git)
Build type: Debug
Branch: main
Hash: 61c1b2bfc2339b272d3fa44bb9dbb6bdba528a64
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-17:
@abdullahtahiriyo would you have 'back of a napkin' proposal on how to implement this ?

xtemp09, 2023-11-17:
_ffmpeg_ always shows [this in logs](https://github.com/FFmpeg/FFmpeg/blob/08e97dae205d10806a0360bfc62f654d629dda93/libavutil/log.c#L385). It just counts messages.

abdullahtahiriyo, 2023-11-22:
Several issues at once.

1) redundancy in the sense that it is reported in the Notification Area and in the Report View

This is intended (*). The Report View is for the developer. The Notification Area is for the user. 

If the user has developer messages redirected to the user, then they will appear in both.

(*) It is intended that if a message is intended for both developer and user (or the redirection is enabled), they appear in both. But if a developer message is appearing in the Notification Area (with the redirection disabled), it means the developer of that module is not using the right call for the error/warning. There are variants for c++ and python, so that developers can decide which function should be called depending on the intended recipient.

In the screenshot, it appears that the message is a developer error/warning, and should not be sent to the user (at all, or at least in that form). This is the chart to decide on the function:

[image]

2) The message list is intended to keep under a limit the number of messages that get **stored** in the notification area **once notified**.

3) It seems quite possible to implement this request (message repeated n times). Let me think about it. (I leave this GH notification unread and will come back).

luzpaz, 2023-12-14:
@xtemp09 can this ticket be closed ?

abdullahtahiriyo, 2023-12-17:
@xtemp09 

Umm, I am not sure the counting is improper.

The Notification Area is global and thus accumulates messages from different documents over the whole FreeCAD session. It is not intended to be "reset" (beyond limits to the overall number of messages). 

One of the reasons is that it eyes to support multi-document assemblies/projects. That is also one of the reasons to introduce the "notifier" string and the advice to make the "notifier", whenever possible, the name of the document or document#object.

A message is equal (and thus accumulated as repeated), when the notification type (error/warning/...) is equal, the notifier string is equal and the message is equal.

I do not think it is reasonably feasible for the Notification Area to keep track of open/closed documents. Even if it did, when the string is not the document or document#documentobject name type, it would be impossible for the Notification Area to know that the originator is a different document or document object.

Taking this into account, I think the behaviour shown by the Notification Area is the intended one: a) the message is notified, b) an identical message is not repeated, c) the number of repetitions takes into account all the equal messages in a row.

With respect to the Report View, the Report View is not reset either when changing document. In the whole list there will be 30 instances (unless they were manually cleaned). The counter should match...

What I realise now is that maybe this issue ticket is to apply the feature to both, the notification area and the report view. Indeed, I did not change the behaviour of the Report View. 

Unlike the Notification Area, currently the Report View does not store messages and is intended to output text as soon as it is received as output only (as if it were a terminal). It is not intended to change already output lines (also because each line has a timestamp). We could store the message and use it to add at the end of a repeated message the number of repeated instances, but we would still have that number of "lines" in the view. Like:

```
13:57:00 my notifier: my message
13:57:05 my notifier: my message (2 times)
13:57:10 my notifier: my message (3 times)
```
However, I am unsure how useful this would actually be...

13:57:10 my notifier: my message (3 times)

xtemp09, 2023-12-17:
Ok. Thank you.

closed 2023-12-17 by xtemp09