FreeCAD/FreeCAD #11184: [Problem] Missing API documentation for `src/Mod/Part/App/ShapeFix/ShapeFix_WirePy.xml`

luzpaz, 2023-10-26:
https://github.com/FreeCAD/FreeCAD/blob/410d4f9f6acb56cf84f1117dd39a595bcd551f15/src/Mod/Part/App/ShapeFix/ShapeFix_WirePy.xml#L127-L136

https://github.com/FreeCAD/FreeCAD/blob/410d4f9f6acb56cf84f1117dd39a595bcd551f15/src/Mod/Part/App/ShapeFix/ShapeFix_WirePy.xml#L163-L167

https://github.com/FreeCAD/FreeCAD/blob/410d4f9f6acb56cf84f1117dd39a595bcd551f15/src/Mod/Part/App/ShapeFix/ShapeFix_WirePy.xml#L178-L182

https://github.com/FreeCAD/FreeCAD/blob/410d4f9f6acb56cf84f1117dd39a595bcd551f15/src/Mod/Part/App/ShapeFix/ShapeFix_WirePy.xml#L225-L361

CC @RohanMishra315

shivam200446, 2023-10-26:
I want to work on it assign me!

RohanMishra315, 2023-10-26:
I'll do it @luzpaz

luzpaz, 2023-12-12:
Thanks @RohanMishra315 ! Closing