FreeCAD/FreeCAD #11189: [Problem] Opening .3mf file results in <class 'RuntimeError'>: Zip file consistency problem. Failure while reading zip file central directory

nmaludy, 2023-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to open an .3mf file (example https://www.printables.com/model/361829-rugged-gridfinity-box/files Lid-4x5.3mf) an error is printed in the Report View:
```
09:52:03  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'RuntimeError'>: Zip file consistency problem. Failure while reading zip file central directory
```

I'm able to open the file in Onshape, so i know the file itself is good.

Note:
- FreeCAD is installed via Flatpak, not sure if that has something to do with it?
- Running Rocky Linux 9

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at f670854)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-10-27:
It is indeed related to the ZIP format. If I create a new zip from the content of the file and give it the 3mf extension, the model can be imported.

wwmayer, 2023-10-29:
Looks like a bug in the zipios++ library. It fails to determine the correct position of the "Central Directory" of the zip file.
Because of this the stream position is set to an invalid position and reading from the file fails.

vkorenev, 2024-02-07:
I'm getting the same error for some of my .3mf files. They are not corrupted. Inspecting them with `file` utility displays the following:
```
Zip archive data, at least v4.5 to extract, compression method=deflate
```
So, probably, zip64 extensions are causing this error.

luzpaz, 2024-03-14:
Is there an upstream ticket for this yet?

wwmayer, 2024-03-15:
This one `https://github.com/Zipios/Zipios/issues/43` sounds very similar.

clement-or, 2024-03-24:
Same here with [this model](https://www.printables.com/fr/model/486979-folding-rule-holder-for-skadis-board/files), the aforementioned workaround didn't fix it for me.

My OS
```
NAME="Pop!_OS"
VERSION="22.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 22.04 LTS"
VERSION_ID="22.04"
```

m10d, 2024-05-09:
blocked by this as well - similar to others, re-compressing the file doesn't fix (although importing after following the same steps in https://github.com/FreeCAD/FreeCAD/issues/11189#issuecomment-2000198671 doesn't load AND does not produce an error.  My model YZ Plate, attached here (postpended `.zip` to file name, so it would upload), from https://www.printables.com/model/204709-lowrider-3-cnc/files
[YZ Plate v1.3mf.zip](https://github.com/FreeCAD/FreeCAD/files/15264586/YZ.Plate.v1.3mf.zip)

Roy-043, 2024-08-12:
To convert the .3mf file you can try the Python code below. Change the `old` and `new` path and then paste in the FreeCAD Python console.

```python
import zipfile

old = "C:\\Downloads\\plate.3mf"
new = "C:\\Downloads\\plate_new.3mf"

old_zip = zipfile.ZipFile(old, "r")
new_zip = zipfile.ZipFile(new, "w")

for name in old_zip.namelist():
    new_zip.writestr(name, old_zip.open(name).read())

old_zip.close()
new_zip.close()
```

closed 2024-11-28 by wwmayer