FreeCAD/FreeCAD #11192: [Problem] Can't edit PropertyLinkSubList from GUI

FlachyJoe, 2023-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When clicking the three dot button of a PropertyLinkSubList the user is able to select (sub)elements but the property doesn't reflect the choice.

### Full version info

```shell
[code]
OS: Debian GNU/Linux trixie/sid (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34672 (Git)
Build type: Release
Branch: modernize-type-check
Hash: aa529507480725a3a73f0d8d7985ef612865345c
Python 3.11.6, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-27:
This is a regression, right ?

FlachyJoe, 2024-05-28:
Problem occurs with PartDesign features but not with Part ones.

PR #14345 (2024-06-10): Fix SubElements selection for PropertyLinkSubList
Fix #11192
closed 2024-06-10 by chennes