FreeCAD/FreeCAD #11200: Link objects error when exporting to STL or DAE

bjnobre, 2023-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've realized any 2nd links to a part on assembly4 don't be exported. See it bellow.

source:
[image]
output:
[image]

I've tried also on every other formats, and the same issue occurs. I've tried with another project, with more simple shapes and the the errors still there.

souce:
[image]
output:
[image]

Even on draft 2d drawings, linked objects are not exported:

source:
[image]

output:
[image]

So, I assume that it's a issue with exporting module. I am using 0.22 dev version, but it occurs on 0.21 version.

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34789 (Git)
Build type: Release
Branch: main
Hash: 41503d63ebc3de96d156e6eb0cdb889ebb150bd4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * Assembly4 0.50.2
  * fasteners 0.4.62
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-27:
Have you reported to Assmebly4 repo ?  
Can you also include the the file (zip it and drag-drop in to gitihub textbox)

Roy-043, 2023-10-28:
No need for a file. The Mesh WB does not export Links.

luzpaz, 2023-10-28:
@Roy-043 thanks. Can you update the ticket title then, if you don't mind ?

Roy-043, 2023-10-29:
This a regression. The problem does not occur in V0.20.
Forum post reporting the same issue:
https://forum.freecad.org/viewtopic.php?t=82289

bjnobre, 2023-10-30:
I've tested an assembly4 export on fc 020 abd on fc 022. I've got the expected result on the older version.

on fc 022
[image]

on fc 020
[image]

the files on both version:
[A4-test-in-fc0200.zip](https://github.com/FreeCAD/FreeCAD/files/13208660/A4-test-in-fc0200.zip)
[A4-test-in-fc0220.zip](https://github.com/FreeCAD/FreeCAD/files/13208661/A4-test-in-fc0220.zip)

Montaire, 2023-11-25:
I have the same issue with the **35176-2023-11-23** development version as well.

In FreeCad it looks like this:
[image]

Exported from 21.1 release:
[image]
From the supports only one side element is exported but nothing more.

Exported from 22  35176-2023-11-23 release:
[image]
Here all the left side elements are exported.

And here is my model, I am not sure if I did anything wrong here:
[battery_holder_v3.6.zip](https://github.com/FreeCAD/FreeCAD/files/13464191/battery_holder_v3.6.zip)
The side element is using mirroring, and its in the same object as the front element.

I may accidentally created the file which has all the problems and symptoms that's possible, "the veterinarian's horse".

bgbsww, 2023-11-25:
See https://github.com/FreeCAD/FreeCAD/issues/10545#issuecomment-1826325891  It is a bug, but not this bug; it exists independently of links.

Montaire, 2023-11-25:
Hey @bgbsww 

Thank you for your feedback.
I created a new version, and it seems removing the elements from the extra object wrapper, and instead putting them into a compound object does help! And also I used a link for the left side of the support, and mirrored that.

[battery_holder_v3.6.zip](https://github.com/FreeCAD/FreeCAD/files/13464508/battery_holder_v3.6.zip)

Now with version **21.1** I get the first support as it should be.
[image]

With the dev version 22 **35176-2023-11-23** the rotation is all around the place, but the core issue is fixed well with your commit!
[image]

bgbsww, 2023-11-25:
Thanks for the additional file, @Montaire !  I haven't found it yet, although I'm suspicious of https://github.com/FreeCAD/FreeCAD/blob/7561b90eb8bdbb0e692651d98d853da605e4deef/src/Mod/Mesh/App/Exporter.cpp#L51, its cache, and its interaction with https://github.com/FreeCAD/FreeCAD/blob/7561b90eb8bdbb0e692651d98d853da605e4deef/src/Mod/Mesh/App/Exporter.cpp#L104-L111 which loops twice when processing "Support" in the original file.

Since the support_face does not export, I'm more suspicious of the mesh content than the looping, but I need to step away and it will probably be a day before I get to digging deeper.

bgbsww, 2023-11-26:
This isn't related to the links or lack thereof, and thus a different bug.  It is related to the precise structure of the "battery_holder_v3_6#Part" labeled as "support".  Creating a new Part with a Body inside and a Mirror does not duplicate the bug, but it also does not have exactly the same structure as the broken one.

I think it may be related to using a list of subobjects as opposed to the Group objects inside the Part, and if this is the case, it's a substantial change to how Exporter::addObject functions.

I'm going to try to duplicate the process of creating a broken Part to narrow this down.

bgbsww, 2023-11-26:
OK!  So, bodies without Tips are not exported, and Mirrors do not export both the source and the mirror.

If you take support_front (the body ) in the example and add a Tip of SupportFront ( the pad ) then the support starts showing up in the export.  I'm not sure this is really a bug, as when I create bodies and place things in them, the Tip is set.
Any idea how this was created?  Maybe there's a bug in there not setting this value.

However, the screwy positioning is related to the fix in this bug, and the latest PR corrects that.

Should we open a separate bug for "Mirrored objects do not appear in exports to STL or DAE"?

And maybe a bug for "Bodies without Tips do not appear in exports to STL or DAE?"  if the consensus is that this is an incorrect behavior?

Montaire, 2023-11-27:
@bgbsww, its a good question, I fiddled around a lot with the support for the resin print. Should I try to replicate how I did it? I can do it in the afternoon. Probably it was introduced by dragging in and out the lofts, but I am not entirely sure. Thank you again for your help!

PR #11539 (2023-11-29): Don't apply the transformations recursively, just make sure the right…
… one is applied to the top subobject.

Fix #11200 Fix #10545   Without nasty side effects as evidenced by @Montaire example files.
closed 2023-11-29 by adrianinsaval