FreeCAD/FreeCAD #11208: [Problem] Sketcher toolbars are cluttered

PaddleStroke, 2023-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sketcher toolbars are overcrowded.

Here I am compiling a list of ideas to improve the situation : 
1 - Make a command group for 'Split', 'Extend' and 'Trim'.
2 - Make a command group called 'Shapes' for 'Rectangle' 'Slot' and 'Polygon. (Note right now we have 3 rectangle tools and many polygons, but with currently opened #11174 we can have only 1 tool for each.)
3 - Rendering order could go inside the setting button of the element widget
4 - Snap command group could go inside the setting button too.
5 - 'Show hide circular helper' : This should not require a tool. I think the underlying issue is that the constraints should render circular helper when needed.
6 - Clone/move/copy/rectangular array could be done in one 'Translate' tool with tool parameters (I've got it ready already)
7 - Coincidence and PointOnObject could be a single tool (ah yes this is a thorny topic).
8 - 'Carbon copy' tool should not require a tool. Copy paste of a sketch should be easier than that. Either just selecting it in the tree and dragging it in the 3dview. Or ctrl-C of the sketch on the tree and ctrl-V in the 3dview...
9 - 'Increase/decrease B-Spline degree' could be made a command group as is 'modify knot multiplicity'

@obelisk79 is that something the Design Working Group want to discuss ?

### Full version info

```shell
0.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2023-10-27:
I would like to review toolbars to optimize use and recently started drafting some guidelines for toolbar/command management. I think an in-depth look at sketcher toolbars would be a good first place to look, however I would like to get the draft guidelines a little better defined so the same process can be applied to other workbenches as well. Perhaps in roughly a week I can bring this to the design working group for debate?

PaddleStroke, 2024-07-02:
Is issue is too broad. Many points in it has been addressed. Rest should have their own issue if needed.

closed 2024-07-02 by PaddleStroke