FreeCAD/FreeCAD #11209: [Problem] PartDesign: Boolean Bodies not Visible while Editing

maxwxyz, 2023-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Often, complex models are build by boolean operations of different bodies. This is possible with the boolean feature in PartDesign.
In the example file [BooleanBodiesBefore.zip](https://github.com/FreeCAD/FreeCAD/files/13191784/BooleanBodiesBefore.zip), I want to 
- Fuse the Body _Subtract2_  to the Body _Subtract1_
- Fillet the joining edge of the newly formed united Body _Subtract1_
- Cut the "complex" Body _Subtract1_ from the _MainBody_

After these Boolean operations, it is not possible to view the "original" Bodies _Subtract1_ and _Subtract2_ if I would like to add or remove features and cannot select references, features, faces or edges of them - they are hidden in the 3D view. I can only manipulate features which are accessible in the tree view (change parameters which are already there). This is possible with other CAD software like CATIA, NX and so on.
Hiding features or setting the tip before, inside or on the first item of one of the nested bodies does not make them visible.

This is the situation before:
[image]

After joining the yellow body (Subtract2) to the red body (Subtract1), filleting the resulting edge, I cut the newly formed body from the green body (MainBody) to get my result:
[image]
[BooleanBodiesAfter.zip](https://github.com/FreeCAD/FreeCAD/files/13191836/BooleanBodiesAfter.zip)

Here is the example file after the boolean operations: [BooleanBodiesAfter.zip](https://github.com/FreeCAD/FreeCAD/files/13191836/BooleanBodiesAfter.zip)

Currently the workaround is to delete the Boolean features and edit the bodies and repeat **all** operations. For example to edit Body _Subtract2_ I delete the nested Boolean feature in Body _Subtract1_ and FreeCAD deletes the following fillet feature. This is very painful in complex models. Another workaround is with shape binders, but this is blowing up the tree view when using multiple bodies. 

Here a quick video of trying to access / see the nested bodies and edit them. Also what happens when deleting the nested Boolean feature:

https://github.com/FreeCAD/FreeCAD/assets/6246609/400cd753-b59e-4253-bf3b-4924089c3c98

In @realthunder Link Daily, there is a workaround with automatic shape binder creation (keeping the tool as separate body) but allowing to edit the features in their respective bodies (and it even keeps the colors of the resulting faces after boolean operations):
[image]

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34789 (Git)
Build type: Release
Branch: main
Hash: 41503d63ebc3de96d156e6eb0cdb889ebb150bd4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.5
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * Defeaturing 1.2.1
  * fasteners 0.4.67
  * freecad.gears 1.0.0
  * freecad_metal_workbench
  * OpenDark 2023.10.20a
  * sheetmetal 0.3.1
  * Silk 0.1.5
[/code]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2023-10-28:
The Link branch workaround is usable in main too but you has to manually create the Binder before use it in the boolean.

maxwxyz, 2024-02-24:
@davesrocketshop would individual face colors or colors for part design features conflict with the overall material of a body?

maxwxyz, 2024-08-30:
Despite the binder workaround it is also possible to manually change the Display parameter in the View properties of the Boolean feature from "Result" to "Tools".
Then the child objects can be edited and are displayed.

Maybe this behavior could be activated by default when the tip is on the boolean or the body/part inside of a boolean is the active body/part.

[image]

Mr-Rahul-Paul, 2026-01-22:
I have an idea , a similar logic from a previous PR
We can add an event listener that automatically changes the visibility mode when a nested body is activated.

we check if the activated object is tool body inside our boolean. If it is, we automatically switch the Boolean's Display property reveal the hidden body for editing 

do you think this is a valid solution @maxwxyz , I believe i can figure it out