FreeCAD/FreeCAD #11210: [Feature Request] Part Design: Keep Body Colors of Resulting Faces after Boolean Operations

maxwxyz, 2023-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When doing boolean operations in Part Design WB with different colored bodies, the resulting shape is always the color of the current body. In CATIA or @realthunder link daily, the resulting faces of the boolean operation have the color from their original body (tool body).

Initial situation:
[image]

Result in FreeCAD:
[image]

Result in RT link:
[image]

### Full version info

```shell
0.22
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-01-05:
Maybe related to https://github.com/FreeCAD/FreeCAD/issues/5696

maxwxyz, 2024-02-24:
@davesrocketshop would individual face colors or colors for part design features conflict with the overall material of a body?

davesrocketshop, 2024-02-24:
> @davesrocketshop would individual face colors or colors for part design features conflict with the overall material of a body?

The part has a material and an appearance. The shape's material will define the default colors, but the appearance can be set separately, as it can now.

Roy-043, 2024-05-06:
Related?: #6069

maxwxyz, 2024-05-07:
@Roy-043 no I don't think so. This one is to apply the colors to the cut faces of the body from the faces of the other body which it was cut by.

maxwxyz, 2026-02-01:
@theosib could you let the AI check this out? This is implemented in the Realthunder fork but with the different material system it is not that easy to port the element map history.