FreeCAD/FreeCAD #11211: [Feature Request] Part Design: Automatically Add Groups in Bodies for Sketches and Datums

maxwxyz, 2023-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When working with features in Part Design which rely on a sketch, the sketch becomes a child of the feature and is hidden. Trying to access the sketch in complex models is hard, as you have to look through every feature in the tree, expand it and check if it the one you need. There is a workaround to create a group outside of the body and keep the sketches there, but they are separated from the body and you have to manually expand the feature and drop the sketch in the external group.
Other CAD software creates automatically groups and stores sketches in the sketches group, datums and axis in a datum group.
I've just learned, that said feature is already implemented in @realthunder daily link:
[image]
Could such a feature be implemented in FreeCAD?

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34789 (Git)
Build type: Release
Branch: main
Hash: 41503d63ebc3de96d156e6eb0cdb889ebb150bd4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.5
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * Defeaturing 1.2.1
  * fasteners 0.4.67
  * freecad.gears 1.0.0
  * freecad_metal_workbench
  * OpenDark 2023.10.20a
  * sheetmetal 0.3.1
  * Silk 0.1.5
[/code]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-03:
@realthunder could you locate this feature, maybe I could give this a try.

realthunder, 2024-03-04:
The feature is initially implemented in [this](https://github.com/realthunder/FreeCAD/commit/c84416d96ef) and [this](https://github.com/realthunder/FreeCAD/commit/2cfc5e80) commit. There are a few small follow up fixes, such as [this](https://github.com/realthunder/FreeCAD/commit/1db27a705), [this](https://github.com/realthunder/FreeCAD/commit/dad2849) and [this](https://github.com/realthunder/FreeCAD/commit/8656e86).

maxwxyz, 2024-03-07:
Thanks, unfortunately this is too much for my current abilities to implement on my own.

maxwxyz, 2024-03-09:
To add from the FreeCAD Day 2024 Complaint Session:
**The tree isn’t actually just a tree its a collection of things used to construct it**
The tree represents chronological things. Sketches wind up down inside the tree in the order things are happening instead of above. The tree isn't actually just a tree, it's also collection of things used to construct it, but it's it mixes things. Putting items down deeper in the tree which makes them hard to find.

PaddleStroke, 2025-07-15:
Have this been discussed by @FreeCAD/design-working-group ?

maxwxyz, 2025-07-15:
@PaddleStroke yes seems like a sensible change

maxwxyz, 2026-02-01:
After some discussion, this feature should not add additional document objects to the document, but just display the existing elements (sketches, datums, binders/references,...) in visual groups/folders on the top of the body.

pierreporte, 2026-02-01:
@maxwxyz Just to be sure, are those groups just repeating what is in the feature tree, like shortcuts, or is everything moved in these categories?