FreeCAD/FreeCAD #11212: [Feature Request] Part Design: Temporarily Suppress / Hide Features

maxwxyz, 2023-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Maybe this is only possible with a solution to the TNP but currently you can only view a current state of a body in Part Design WB by pressing the space bar on a feature, hiding all following features or set the tip to said feature.
In other software and in @realthunder link daily there are features to temporarily hide or suppress multiple features in any stage of the bodies feature tree:
[image]

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34789 (Git)
Build type: Release
Branch: main
Hash: 41503d63ebc3de96d156e6eb0cdb889ebb150bd4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.5
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * Defeaturing 1.2.1
  * fasteners 0.4.67
  * freecad.gears 1.0.0
  * freecad_metal_workbench
  * OpenDark 2023.10.20a
  * sheetmetal 0.3.1
  * Silk 0.1.5
[/code]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-17:
@FlachyJoe can I close this as https://github.com/FreeCAD/FreeCAD/pull/12096 was merged?

FlachyJoe, 2024-02-20:
@maxwxyz The feature is not fully usable yet because it's not implemented in Gui but feel free to close if you think it's good.

@furgo16 
> there is a missing part of adding some sort of UI to more easily use it

Yes, the feature is hidden until the TNP fix was fully merged due to the problems it can introduce in the model.
I had an idea to improve #12412, I'm working on it.

furgo16, 2024-02-21:
Thanks. Related: https://github.com/FreeCAD/FreeCAD/issues/12361

maxwxyz, 2024-02-21:
Duplicate of #12361

maxwxyz, 2024-03-09:
Reopen to add it in the UI (menu + context menu) when TNP is fixed.

maxwxyz, 2024-05-17:
oh yes, this would be amazing in 1.0!

FlachyJoe, 2024-05-17:
- [x] Unhide the property
- [ ] Add a main menu command to toggle the property
- [x] Add a tree context menu command to toggle the property

PR #14112 (2024-05-20): PartDesign: hide Suppressed property with FC_USE_TNP_FIX flag
fix #11212 

Suppressed is shown when compiled with FC_USE_TNP_FIX
closed 2024-05-20 by chennes