FreeCAD/FreeCAD #11214: [Feature Request] Part Design: Preview of Features in 3D View

maxwxyz, 2023-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be great to have an outlined / transparent overlay as a preview of the feature in part design on top of the previous feature / state of the body. This could help to review any feature in PD before applying it and helps adjusting the right parameters. The benefit of the transparent overlay is also that you see around and through the model / body state before the feature.
[image]
[image]

### Full version info

```shell
0.22
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DrD4ffy, 2024-01-12:
I think this would be very welcome for two reasons:

1. What @maxwxyz mentioned, but not for simple adjustment. The PartDesign-Pad is already visualized while you edit it ... but only if you did it right. If you got confused about your axis (picking X wher you should have picked Y) or the direction (looking at your sketch from the wrong direction) you might pad into the void, create a new solid ... and see nothing. The proposed preview (as done by RT also I think) could always show the user what he is doing and help understand his mistake.
2. It is a matter of consistency. Other tools (e.g. Primitives, Loft, Pipe, ...) already show a preview, even if the operation would end in an error (multiple solids) or simply do nothing because there is no overlapping between the subtracting shape and the existing body.

maxwxyz, 2024-03-29:
Same for e.g. other Part Design features, like datum planes. All elements should be visible as a preview.

martinproks, 2024-04-28:
Hi,
I'm solving similar problem in cosmetic thread 3D. 1st step of PartDesign python feature creation is attachment with direction. And for direction I create temporary object as Draw.makeWire. This feature could be inserted into PartDesign body without problems and later deleted. It can help to visually identify the correct orientation during attachment phase.
[image]

maxwxyz, 2024-04-28:
@martinproks the feature described in the issue is present in @realthunder branch, maybe the code could be used as a baseline?

pierreporte, 2024-04-28:
Something that could be interesting besides preview of future geometry is an API to draw things in the 3D view like arrows, text, custom shapes or thing like that. They wouldn’t be part of the geometry and could have constant dimensions regardless of zoom level. They could even be interacted with. For @martinproks, arrows would be very useful for example. @maxwxyz the idea is similar to what Catia does (arrows of course but also planes you can drag for extrusions or dimensions you can edit on the fly), and Solidworks as well though slightly differently.

pierreporte, 2024-04-28:
@realthunder can this feature be isolated in your branch?

realthunder, 2024-04-28:
It's not easy. Some preview, like pattern feature, requires significant code modification of the feature itself. Some relies on topo naming. I'll see what I can do.

maxwxyz, 2024-04-29:
@pierreporte how is #13116 related to this issue?

pierreporte, 2024-04-29:
@maxwxyz Annotations in the form of text and other kind of geometry (I will comment on #13116 about this) can help better understand what’s going on with the future geometry that is suggested by the preview feature. For example, arrows can be used for normals. Text for dimensions. Better if they can be interacted with, like dragging the end of the extrusion to change the value.

[image]

maxwxyz, 2024-04-29:
thanks for clarification. I guess the annotation labels are not well suited for this task as they are currently implemented. Maybe better would be an interactive arrow where you can pick and pull the pad/thickness/whatever and get real time feedback in the 3D view and display the length, maybe even as input box like the OVP (on-view-parameters) for the tools in Sketcher.
This would be the related issue, you can add your remarks there:
- https://github.com/FreeCAD/FreeCAD/issues/6204

kadet1090, 2025-08-08:
there are still few minor things missing for PartDesign - @maxwxyz was going to make sub-issues for them

PR #23062 (2025-08-31): PartDesign: Implement preview for Booleans
That PR implements missing preview for booleans in PartDesign. It depends on #23019 so should be merged after.

Closes: #11214 

https://github.com/user-attachments/assets/61bad5dc-ab21-4115-a9ad-309768584d5a

## Issues

## Before and After Images
closed 2025-08-31 by chennes