FreeCAD/FreeCAD #11215: [Feature Request] Temporary Visibility of Selected Tree Feature in 3D View

maxwxyz, 2023-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be beneficial to make the current selected tree feature temporarily visible in the 3D view, as long as the feature is selected and no other dialog is open, even when the selected feature is hidden.
It allows for quickly cycling through the tree items without actively showing and then hiding the features (e.g. when looking for sketches, inspecting what a feature does in Part Design Workflow, looking for bodies,...). It is also handy when you are looking for references or creating binders, as the currently selected item (in tree view) is visible in the 3D view even when the body or feature itself is hidden and features of the temporarily visible item can be selected (faces of bodies, edges of sketches or bodies,...).
Demo from RT link:
 

https://github.com/FreeCAD/FreeCAD/assets/6246609/0a1a2524-a394-4e75-96c9-1140b032ea5a

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

levisyang, 2023-10-28:
Well,indeed this is a quite usefull function in SolidWorks.I agree with you!

maxwxyz, 2024-06-28:
@bgbsww this should now be feasible with TNP fixed?

bgbsww, 2024-06-28:
I'm not sure I see the TNP relationship here.  This just looks like a non-TNP feature in the RT branch for someone to look into if they wish.  Or I'm missing something.

maxwxyz, 2024-08-30:
@realthunder how much effort would this be to port it?