FreeCAD/FreeCAD #11221: circle-to-circle constraint does not work when the circles are overlapping

pierreporte, 2023-10-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It is impossible to define a circle-to-circle constraint when the two circles are overlapping, because the distance is considered negative.

Steps to reproduce:

1. Draw two circles, making sure they overlap.
2. Select the two circles.
3. Activate `Sketcher_ConstrainDistance`.
4. Click OK in the dialog box, that should show a negative value. The sketch shows a negative dimensions.
5. The dimension disappears and the following errormessage is written in the report panel.

```
<class 'ValueError'>: Negative datum values are not valid for the constraint with index 0
21:46:59  App.getDocument('Unnamed').getObject('Sketch').setDatum(0,App.Units.Quantity('-4.790000 mm'))
```

There is an alternative way:

1. Draw two circles that *do not* overlap.
2. Select the two circles.
3. Activate `Sketcher_ConstrainDistance`.
4. Click OK in the dialog box.
5. Double-click on the dimension.
6. Write a negative value.
7. Click OK.
8. The dimension disappears and the same error message is shown.

The behavior is different with reference dimensions because they can be created with a negative value. However, it is impossible to switch them to regular dimensions. It then shows again the presented error message.

### Full version info

```shell
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Render 2023.8.18
```

### Subproject(s) affected?

Sketcher

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-29:
Please test with development version https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

pierreporte, 2023-10-29:
The last weekly version has the same bug.

FlachyJoe, 2023-10-29:
Hi, mathematically the distance between 2 overlapping circle is null. IMHO the reference dimension shouldn't be allowed.

pierreporte, 2023-10-29:
@FlachyJoe This dimensioning is necessary for drawing lenses or similar objects. The workaround is to draw a segment between the centers and points at the intersection with the circle. This is actually the only way for arcs at the moment while #9166 is not merged.

The problem may not have sense mathematically but mechanical design is a bit different. I believe that the intent is clear. FreeCAD seems to understand it too but just refuses negative dimensions. If this dimension was made positive, some way or another, it may work directly as intended.

pierreporte, 2023-12-19:
Fixed in the latest weekly:

```
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35414 (Git) AppImage
Build type: Release
Branch: main
Hash: 759930e3aab711355d2f0895f641a8657897af48
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * OpenDark 2023.12.1
```

pierreporte, 2024-01-03:
Reopening since @luzpaz assigned someone.

luzpaz, 2024-01-15:
Assigned to flachy to give credit for the fix. Closing.

closed 2024-01-15 by luzpaz