FreeCAD/FreeCAD #11224: dxf export broken for sketches on non xy-plane

mschuldt, 2023-10-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Dxf export (Autodesk DXF 2D) of a sketch only works when the sketch is on the xy plane.

The minimal example attached to this issue is a sketch on the xz plane of circle inside of a square. The exported dxf has the circle on the xy plane instead of the xz plane. I'm not sure if this problem is only limited to circles. My actual designs are too cluttered to tell what exactly is wrong with the export.

The attached images show the sketch and the exported dxf which has been imported into freecad for viewing. Other programs such as Inkscape show the invalid dxf file in same way.

I have found various ways to partially work around this issue including using the legacy exporter or coping the sketch and then rotating it onto the xy plane. Between these two methods I can export most of my designs but still there are some that are also broken with the legacy exporter and some that cannot be rotated without breaking references.

I find this makes FreeCad practically unusable for creating complicated interlocking designs for processes like laser cutting.

### Full version info

```shell
[code]
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.0.0
  * workfeature
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[dxf_export_bug.zip](https://github.com/FreeCAD/FreeCAD/files/13197116/dxf_export_bug.zip)

The sketch:
[image]

The resulting dxf export:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-10-29:
To avoid breaking models you can create a PartDesign_ShapeBinder and rotate and export that.

mschuldt, 2023-10-29:
I can't find a way to rotate a PartDesign_ShapeBinder, but that workaround does work for me if using PartDesign_**Sub**ShapeBinder instead.

Roy-043, 2023-11-01:
PartDesign_ShapeBinders are more convenient here. But their Placement property is hidden.
See:
https://forum.freecad.org/viewtopic.php?p=716186#p716186

WandererFan, 2023-12-29:
This applies to the c++ dxf exporter in the Import module.

When exporting a sketch to dxf, if the sketch is not on the XY plane, the
result is not correct. Lines and curves that should be connected have gaps.

The exporter code in dxf.cpp (based on HeeksCad exporter) makes assumptions that
the objects to be exported lie on the XY plane.

In order to produce a valid dxf, the sketch should be projected onto its base
plane first. This is essentially what happens when TechDraw exports a sketch, and
these exports are correct when viewed in LibreCAD.

This issue was first discussed in the Polish forum here:
https://forum.freecad.org/viewtopic.php?style=4&t=83203
and here:
https://forum.freecad.org/viewtopic.php?style=4&t=82640

The project first approach could be applied to non-sketch objects, but it is
not clear how to determine a projection direction in this case.

WandererFan, 2023-12-29:
test files. remove the zip.
[test+potek74.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13790326/test%2Bpotek74.FCStd.zip)
[test+potek74_mwe.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13790327/test%2Bpotek74_mwe.FCStd.zip)

WandererFan, 2024-01-02:
I can export the sketch in two orientations.  If I apply the sketch's placement,  the dxf looks like the sketch in edit mode.  If I only use the placement's axis for the projection direction, but don't apply the placement to the sketch shape, the dxf looks like the sketch in display mode.

I'm thinking the dxf should match the sketch in display mode, but I'd defer to strong opinions.

Roy-043, 2024-01-03:
What is display mode? Does it include the rotation of the 3D view?

What is the approach if multiple sketches are selected (which may or may not be co-planar)?

WandererFan, 2024-01-03:
display mode = sketch not in edit viewed in 3d window.

multiple selected objects (sketches) are exported individually using each the normal of each sketch.

Roy-043, 2024-01-27:
Closing this as fixed.

closed 2024-01-27 by Roy-043