FreeCAD/FreeCAD #11230: [Problem] Shortcut for Sketcher CompConstrainRadDia not working

PLChris, 2023-10-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Since the introduction of the smart dimensioning tool the shortcut R for CompConstrainRadDia doesn't work.
When Preferences->Dimensioning is set to both, and the CompConstrainRadDia tool is selected, then the Shortcut should be R as indicated in the help text. However, R always invokes the radius.

### Full version info

```shell
[code]
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34789 (Git)
Build type: Release
Branch: main
Hash: 41503d63ebc3de96d156e6eb0cdb889ebb150bd4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fcgear 1.0.0
  * fasteners 0.4.59
  * sheetmetal 0.2.62
  * ExplodedAssembly
  * Curves 0.6.8
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-30:
@PaddleStroke do you know who worked on smart/auto dimensioning in Sketcher ?

wwmayer, 2023-10-31:
I cannot reproduce this problem either.

luzpaz, 2023-11-15:
@PLChris can you please respond?

Roy-043, 2024-01-14:
I can't reproduce either. Closing this.

closed 2024-01-14 by Roy-043