FreeCAD/FreeCAD #11232: [Problem] Windows, Freecad 0.21.1 -> Please change description of "C:\Program Files\FreeCAD 0.21\Uninstall-FreeCAD.exe"

Saxomania, 2023-10-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

the description of "C:\Program Files\FreeCAD 0.21\Uninstall-FreeCAD.exe" is "FreeCAD - Your Own 3D Parametric Modeler"

thats bad, it should be something like "Uninstalls Freecad"

we had two customers who associated STP files with this EXE (idk why) and as supporter you dont get it on the first view because you see not the associated EXE but their description

btw the description of "C:\Program Files\FreeCAD 0.21\bin\FreeCAD.exe" is "FreeCAD main executable"

so its really hard to differ if only description is seen

### Full version info

```shell
0.21.1
```

### Subproject(s) affected?

None

### Anything else?

https://forum.freecad.org/viewtopic.php?t=82234

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-30:
There are some efforts to fix windows build issues in #10971 and #10337  
Maybe those devs can take a look at this ?

@Saxomania which windows builds are you using ? (Full [about](https://wiki.freecad.org/About) info please)

Saxomania, 2023-10-30:
[code]
OS: Windows 10 build 17763
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]

chennes, 2023-10-30:
@adrianinsaval can you take a look at the language strings for the Windows Installer and see if you can figure out why this particular string is being used in the uninstaller? It looks to me like there's a relevant string commented out.

alexbeer2048, 2023-11-03:
@chennes according to this https://nsis-dev.github.io/NSIS-Forums/html/t-284330.html and that https://nsis-dev.github.io/NSIS-Forums/html/t-310832.html it's not trivial.

Maybe we pick a _FileDescription_ that works for installer and uninstaller. It's a bit of a compromise but should be a good enough solution.

Instead of **"FreeCAD - Your Own 3D Parametric Modeler"** (which could be the description of the main executable)
Something like:
**"FreeCAD Installer/Uninstaller"**
**"FreeCAD Setup/Uninstall"**
or just
**"Setup/Uninstall"** (default for Inno Setup uninstaller files)

closed 2023-11-14 by adrianinsaval