FreeCAD/FreeCAD #11234: [Feature request] Align Camera View to Selected Reference (Face, Datum)

maxwxyz, 2023-10-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are a few buttons in the toolbar for selecting base views, IMHO they could be grouped under the isometric view icon as they are also accessible with the number keys (0,1,...). An example for such a group would be the selection filters in 0.22.
What is missing (and is available in most CAD software) is to align the camera view to a selected reference in the 3D view, with a dedicated button in the core toolbar.
[image]

In the sketcher there is a button to align the 3D view to the plane of the sketch, so why not allow this as a core function in FreeCAD. It would be helpful in many WB, to make splits in Curves WB (always from the viewing angle), activate clipping planes, ...

There is a macro for this I have found in the Addon Manager "Align View to Face" https://wiki.freecadweb.org/Macro_Align_View_to_Face which works good for faces and datums but I have to install it on every machine and create a custom button for it. Would this be a start to merge this function and eventually update its features?

The new toolbar could look like this (obviously with an own icon for the align view button):
[image]

The align view to reference tool should align the camera to the previous selection:

- Faces and Datums > Normal view (like the macro does today)
- If the camera view already matches the selection, maybe reverse it to view from behind (- normal direction) or a button (even in dropdown menu of the new button) to reverse it.
- Allow the selection of edges, lines and axes and set their direction to the normal position of the camera view

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34810 (Git)
Build type: Release
Branch: main
Hash: 196cbecde66a1c3a1b56b8c532ad1faedff4818b
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.7
  * CurvedShapes 1.0.4
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.68
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.1
  * Silk 0.1.5
```
```

### Subproject(s) affected?

None

### Anything else?

https://wiki.freecadweb.org/Macro_Align_View_to_Face

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-10-30:
@PaddleStroke you reworked already some toolbars for the better - what is your opinion on this?

PaddleStroke, 2023-10-30:
I think you should split your issue in 2. Because its really 2 separate topics.
On the grouping of standard views, I think many people will disagree.

For the align to face tool, I think indeed that this is a missing feature that should be in core by default.
Note: If this tool is implemented, the sketcher tool could be removed as they would be redundant.

maxwxyz, 2023-11-11:
I think it would also be great, if the view direction (camera) can be aligned to a selected clipping plane (normal)

FreeCutter, 2024-03-19:
According to here [https://forum.freecad.org/viewtopic.php?p=718524#p718524](url) it should be already be available in Realthunder's version. I can't tell if this correct - thus please double check by yourselfe.

maxwxyz, 2024-03-19:
@FreeCutter it is there but it does not work for me. Its sometimes the opposite and not working for edges or planes

maxwxyz, 2024-04-21:
@Rexbas you have great experience with navigation controls, camera movement and animations: is this something you could get done for the v1.0 feature freeze in mid May? I can provide an icon for the command and add the command to the toolbar.

jwueller, 2024-04-21:
Would this also fix the janky camera jumps when trying to focus a sketch, or should I create a separate issue for that? It seems that the camera orientation doesn't always animate from where it is actually looking, but some arbitrary other orientation that I'm not sure where it comes from.

Rexbas, 2024-04-21:
@maxwxyz, I will try my best :) 

@jwueller, I think I know what you mean but I am not sure. Better make a new issue, preferably with video.

maxwxyz, 2024-04-21:
nice!!

Rexbas, 2024-04-28:
I don't think this will be finished before mid may, I have trouble creating a Python script while doing https://wiki.freecad.org/Line_drawing_function#Testing_the_script. I get `ModuleNotFoundError: No module named 'exercise'`

maxwxyz, 2024-04-28:
does it need to be in python?
any chance a stripped down version could be implemented, e.g. just faces?

this is a sometimes working version in RT 
https://github.com/realthunder/FreeCAD/commit/b6bd57a26effaaf3a3560a4656815080cc92f29f

https://github.com/realthunder/FreeCAD/blob/07c85fedeae51f76d1999f15b24d3c921f623f8d/src/Gui/CommandView.cpp#L1729
https://github.com/realthunder/FreeCAD/blob/07c85fedeae51f76d1999f15b24d3c921f623f8d/src/Gui/View3DInventorViewer.cpp#L4322

Rexbas, 2024-04-28:
I managed to run a Python script by placing it in my build directory / Mod.

There is a `face.normalAt(...)` like used here: https://wiki.freecad.org/Macro_Align_View_to_Face. I am not aware of a `normalAt` method in C++ but if there is let me know.

I tried the code in RT but I could not get it to work. It always goes to the same orientation for a given body no matter which face is selected.

I see this feature request as two problems
1. Getting a (normal) vector from whatever the situation is (face selection, edge selection, sketch plane, ...)
2. Aligning the camera with the obtained vector using the smallest possible angle.

I think I can do point 2 relatively quickly as I have done something similar for the navigation cube. However, point 1 will take more time. I am still figuring out how to do this. I will continue working on this when I have time but probably won't be finished before feature freeze. Suggestions are welcome.

maxwxyz, 2024-04-28:
I'm no expert.
for 1. if selection is

- one line, edge or datum axis just use the direction / vector as input
- is a plane: selectionofplaneobject.Axis().Direction() maybe
- is multiple lines, maybe some average plane could be calculated
- is a face, maybe something like: BRepGProp::SurfaceProperties(face, point, normal) exists or it could be calculated with the UV bounds (https://dev.opencascade.org/node/72634#comment-2320) 

and if the camera is already oriented this way, flip it to view it from the other direction.

@PaddleStroke do you know how from your work in sketcher, assembly or with the offset tool know how to get the normal vector in C++ for selected face(s)?

Rexbas, 2024-05-08:
I made a simplified version of RT code that works for faces and edges. I will make a draft PR tomorrow.

maxwxyz, 2024-05-08:
So cool! Thanks.

Rexbas, 2024-05-08:
I am currently trying to make a command. Would be nice if you could make the icon and place it in the right spot after I created the draft PR.

maxwxyz, 2024-05-08:
sure, just tag me in the PR.

maxwxyz, 2024-05-21:
@Rexbas I've assigned you to continue this great work

- [x] One Face
- [x] One Line
- [x] Datums (axis/plane)
- [ ] If one line is a curve or circle and laying in one dimension / plane --> use this normal direction
- [ ] If the camera view already matches the selection, maybe reverse it to view from behind (- normal direction) or a button (even in dropdown menu of the new button) to reverse it.
- [ ] Multiple faces (average direction)
- [ ] Multiple lines (if they lay on one common plane (always with selection of 2 straight lines), align to this plane's normal)
- [ ] Allow the view to be aligned to current clipping planes
- [ ] Replace ViewSketch with the new tool and set it to the active sketch plane https://wiki.freecad.org/Sketcher_ViewSketch

maxwxyz, 2024-05-30:
@Rexbas I just figured out it works for PartDesign datums but not for the origin objects (default axes and planes for bodies and part objects)
[image]
I've added them to the post above

FEA-eng, 2024-06-13:
The feature itself is merged. If the issue is supposed to stay until its missing capabilities are added then maybe the issue should be renamed or split to make it clear that it’s about improving the existing tool.

maxwxyz, 2024-08-02:
@Rexbas not sure if relevant but the Draft working plane actually is also able to be set to the selection and it works by selecting a face, three points, two lines, and a profile which is on one plane. Maybe some code could be reused?

luzpaz, 2025-05-11:
Is this technically a 'Navigation' issue ?

maxwxyz, 2025-05-12:
I dont think so