FreeCAD/FreeCAD #11235: [Problem]

M4xTheMan, 2023-10-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**How it behaves now**
While using the Constraint contextual dimension in Sketcher WB, the text which belongs to the constraint follows the cursor while working with the tool. Currently, the dimension and the cursor overlap. 

**How it looks now**
[image]

**How it should behave**
It would be nice, if the dimension would get a kind of offset in a way that the dimension and the cursor don't overlap anymore.

### Full version info

```shell
[code]
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34367 (Git) AppImage
Build type: Release
Branch: master
Hash: e36e16de59dece2d67caf478d25e3a6b825102c6
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * LCInterlocking
  * kicadStepUpMod 10.19.5
  * Lithophane
  * lattice2 1.0.0
  * Defeaturing 1.2.0
  * Manipulator
  * fasteners 0.4.66
  * Assembly3 0.12.0
  * sheetmetal 0.3.0
  * Assembly4
  * Curves 0.6.13
  * POV-Ray-Rendering
  * 3DfindIT 1.2.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://forum.freecad.org/viewtopic.php?p=715787&sid=be14851e59aea28c1ecb7270a4664e0d#p715787

@PaddleStroke - ping as you've asked for at the forum

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

M4xTheMan, 2025-01-28:
Still relevant.

`OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * LCInterlocking
  * kicadStepUpMod 11.5.2
  * Lithophane
  * lattice2 1.0.0
  * Defeaturing 1.2.3
  * Manipulator
  * fasteners 0.5.35
  * Assembly3 0.12.2
  * sheetmetal 0.7.3
  * InventorLoader 1.5.1
  * Assembly4
  * Curves 0.6.53
  * POV-Ray-Rendering
  * OpenDark 2024.3.13
  * Gridfinity 0.8.0
  * Freecad-Built-in-themes-beta 1.2.2
`

FEA-eng, 2025-06-08:
Still relevant

closed 2025-07-20 by kadet1090